OpenNeptune3D/OpenNept4une #2: QUSTION:using screen?

Hoorow, 2023-12-06:
screen uses uart protocol to comminucate with printer so if we change screen firmware and make custom firmware for screen to use it with printer.
i cant remember the website for screen details.

breaki1904, 2023-12-08:
You mean https://r6e.github.io/neptune_4/hardware.html ?

halfmanbear, 2023-12-08:
Yes to test you can connect the screen and use the following.

sudo stty -F /dev/ttyS1 115200

printf "page 1\xFF\xFF\xFF" > /dev/ttyS1

printf "page 11\xFF\xFF\xFF" > /dev/ttyS1

Etc

Should change what page is displayed on the screen.

Sineiko, 2023-12-08:
The Settings menu appears on the display. It also makes a beeping sound when clicked. But it does not react to what I click on.

ringnutz, 2023-12-09:
@halfmanbear  is there a current effort to get the screen working with the stock Elegoo firmware? I assume it should be possible

halfmanbear, 2023-12-10:
Yes, it is possible to interface the screen with this, custom firmware might well be an easier route for the screen. 

Would need to write an interface between Klippers API & the screens UART on /dev/ttyS1

cvopicka, 2023-12-11:
Generic firmware that could be compiled like Klipper does for every machine but for varying displays would be the most versatile.  (I didn't say easy).  I also have a non-Elegoo that I put Klipper on and lost function of the display.  It would be nice to solve it in one place universally.  The N4Pro is higher on my list to resolve this for.

Would there be any benefit to seeing how KlipperScreen deals with the Klipper API? I understand the UI is different than what I believe these screens can handle, but it still must do I/O with Klipper to work.  

I am unfortunately time poor right now but would love to participate if possible.  I am not suggesting that anyone here even take this on as it is not a small project idea.  Just my thoughts on the ultimate solution.

f-garofalo, 2023-12-12:
I created a simple Python script to test the Display, it's just an experiment.
It needs PySerial to work

```
#!/usr/bin/env python

import serial, time

SERIALPORT = "/dev/ttyS1"
BAUDRATE = 115200

ser = serial.Serial(SERIALPORT, BAUDRATE)

ser.bytesize = serial.EIGHTBITS #number of bits per bytes
ser.parity = serial.PARITY_NONE #set parity check: no parity
ser.stopbits = serial.STOPBITS_ONE #number of stop bits
#ser.timeout = None          #block read
#ser.timeout = 0             #non-block read
ser.timeout = 2              #timeout block read
ser.xonxoff = False     #disable software flow control
ser.rtscts = False     #disable hardware (RTS/CTS) flow control
ser.dsrdtr = False       #disable hardware (DSR/DTR) flow control
ser.writeTimeout = 0     #timeout for write

print('Starting Up...')

try:
    ser.close()
    ser.open()

except Exception as e:
    print ("Error open serial port: " + str(e))
    exit()

if ser.isOpen():

    try:
        ser.flushInput() #flush input buffer, discarding all its contents
        ser.flushOutput() #flush output buffer, aborting current output

        while True:
            print("Insert command (page 1, page 2, ...) for Elegoo Display: (Ctrl-C for exit)")
            data = input()
            padding = [0xFF, 0xFF, 0xFF]
            ser.write(str.encode(data))
            ser.write(serial.to_bytes(padding))
            print("Write data: " + data)
            time.sleep(0.1)

            print("Waiting for user interaction from Display (Ctrl-C for skip)")
            readData = []
            try:
                while True:
                    response = ser.read()
                    if response is not None and len(response) > 0:
                        readData.append(response.hex())
                    if len(readData) == 6:
                        print("Data read from Display: ")
                        print(readData)
                        break
                    else:
                        time.sleep(0.2)
            except KeyboardInterrupt:
                print("\n")

    except Exception as e:
        print("Error communicating...: " + str(e))
    finally:
        ser.close()

else:
    print("Cannot open serial port!")
```

f-garofalo, 2023-12-12:
If you don't want install pyserial, i have a zip with all dependencies.

Unzip and copy on neptune 4 home folder
```
cd neptune4-display-test
python3 display.py
```

[neptune4-display-test.zip](https://github.com/halfmanbear/OpenNept4une/files/13646728/neptune4-display-test.zip)

halfmanbear, 2023-12-12:
awesome, want me to add this to a screen sub dir so people can contribute and hopefully get something functioning?

ringnutz, 2023-12-12:
Should it be it's own repo? Not sure how common the screen hardware is

f-garofalo, 2023-12-12:
> awesome, want me to add this to a screen sub dir so people can contribute and hopefully get something functioning?

Thanks, 
absolutely yes, you can change it as you prefer! :-D

Hoorow, 2023-12-15:
> ###
> [Dec 8, 2023](#issuecomment-1846865546)

yeah im really sorry for not replying, i was sick for 1 week after posting it

Hoorow, 2023-12-15:
> Yes to test you can connect the screen and use the following.
> 
> sudo stty -F /dev/ttyS1 115200
> 
> printf "page 1\xFF\xFF\xFF" > /dev/ttyS1
> 
> printf "page 11\xFF\xFF\xFF" > /dev/ttyS1
> 
> Etc
> 
> Should change what page is displayed on the screen.

thats really good can we compile screen firmware for it?

Hoorow, 2023-12-15:
> awesome, want me to add this to a screen sub dir so people can contribute and hopefully get something functioning?

hell yeah

halfmanbear, 2023-12-19:
ive updated the script in /display it now navigates the pages somewhat and also interfaces with moonraker via (below)
requires https://github.com/alchemyEngine/MoonrakerPy

I have made a lookup table for the page navigation [response_actions.py](https://github.com/halfmanbear/OpenNept4une/blob/main/display/response_actions.py)

its not an optimal layout but its a start

SHexplorer, 2023-12-21:
Hey, I started building a program for the touchscreen myself and collected some infos. I found no more time to develop it further but I like to contribute it here, hope it helps in some way. The archive contains captured serial communication, collected infos about the different pages and a main program with different functions (printing, displaying thumbnail, etc.). Feel free to use it. When you got questions, just ask :)

[touchscreen-connector.zip](https://github.com/halfmanbear/OpenNept4une/files/13744677/touchscreen-connector.zip)

halfmanbear, 2023-12-21:
Awesome thankyou

halfmanbear, 2023-12-24:
> Interesting project....thanks. They only thing holding me off is the lack of screensupport. But what if you buy a Mks pi35ts screen and plug that into the motherboard?? Or is that too simple thinking?

The ZNP motherboard doesn't have the required connection for that screen unless you break out the SPI and reconfigure the chip select?

You could use an old android phone and klipperscreen or just use the mobileraker app over WiFi as a screen alternative.

khusmann, 2023-12-25:
Thanks for the tip, I'm enjoying mobileraker over Wifi now... This might be worth more explicitly mentioning in the docs for newbs like me -- I notice you already have a shoutout in the credits but maybe something like:

> NOTE: The touch-screen will not be functional after this! (Use android+klipperscreen or mobileraker instead!)

I got in a weird situation the other day you might have a suggestion for -- I was an idiot and reset my router while the hot end was hot and then the printer didn't automatically re-connect to the network. I didn't want to power cycle with it still hot (which would briefly shut the fans down, prob not a big deal but i wanted to be safe), so I was a little stuck. Ultimately I removed and re-inserted the wifi dongle which got it back online... But it made me want to have a plan for the future. Here were some things I was thinking:

1) I can always connect it directly to the router via ethernet, but this is a pain because the printer lives far from the router
2) Maybe I should make myself a little physical "e-stop" button I could connect by USB that would stop all motors and set temps to 0?
3) Or is it possible to get a serial terminal into the pi somehow thru USB?
4) Other ideas of what to do when network connection is lost while running headless?

halfmanbear, 2023-12-25:
Yes good idea I'll add details for mobileraker on the touchscreen note.

You can connect via serial on the usbc-port but a physical button would be cool.

Following might have helped possibly,

sudo nmcli device set wlan0 autoconnect yes

khusmann, 2023-12-25:
Awesome, thx. Good to know about the serial on the usb-c port, that's perfect.

Might still make a little physical e-stop + temp display at some point -- will share here if I do.

Thx for all your work on this -- honestly it's what convinced me to get the neptune!

closed 2024-01-31 by halfmanbear