OpenNeptune3D/OpenNept4une #13: Disable USB Storage AutoMount

Justdigit, 2024-01-04:
Sorry, not really an Issue....

But i want to use klipperscreen on an android phone/tablet. And its communicating by USB. As the tutorial here (https://www.youtube.com/watch?v=TcgTrkA8Oj0&ab_channel=CrimsonRepair)

For as now it is not working and i think its has to do with the USB. When i installed in enabeld it.
How to disable?

And is there some where a discord or reddit page for questions?

Thanks for the great software....

Sineiko, 2024-01-04:
discord: https://discord.com/channels/969282195552346202/1153604550855512095

halfmanbear, 2024-01-05:
I don't believe you need to delete the usb automount rule to get adb / klipperscreen working.

I've created a script to make the process of "mounting" adb devices easier. Let me know if this sorts out your klipperscreen.

Run the main script to update the repo then try the following 

chmod +x ~/OpenNept4une/img-config/adb-automount.sh && ~/OpenNept4une/img-config/adb-automount.sh

Justdigit, 2024-01-05:
> I don't believe you need to delete the usb automount rule to get adb / klipperscreen working.
>
> I've created a script to make the process of "mounting" adb devices easier. Let me know if this sorts [...]

You can close this automount has nothing to do with klipperscreen Android.
Told the solution in discord. Didn't tried your solution because it was already working.
Thanks.