OpenNeptune3D/OpenNept4une #15: MCU Flash does not persists power outages

PapaSchlumpf70, 2024-01-05:
I've flashed the mcu according to the manual and everything is working afterward even after switching off the printer via the power switch, but once I pull the power cord the printer does start up, but Klpper seems not to be able to connect to MCU.
I've also tried to fash the provided firmware from GIT and the stored stock firmware without success. Fashing is finalizing fine, but No connection possible from klipper.
Anything I can do here?

halfmanbear, 2024-01-05:
You could try a full wipe 

dd if=/dev/zero bs=1 count=262144 | tr "\000" "\377" > empty.bin

stm32flash -w empty.bin -v -g 0x0 /dev/ttyS0

Then try flash again

PapaSchlumpf70, 2024-01-05:
> You could try a full wipe
> 
> dd if=/dev/zero bs=1 count=262144 | tr "\000" "\377" > empty.bin
> 
> stm32flash -w empty.bin -v -g 0x0 /dev/ttyS0
> 
> Then try flash again

Unfortunately, this did not help. Wiping MCU and flashing the created firmware did always result in no connectivity.  I had to flash the firmware provided here and afterwards the created firmware from kiauh. After starting Klipper everything was working. Rebooting did work fine as well, but once I switch off the printer klipper can't connect to MCU anymore :-(

halfmanbear, 2024-01-06:
What method are you using? The main or alt-method?

if you wipe the flash and do a 

stm32flash -w /home/mks/klipper/out/klipper.bin -v -g 0x8008000 /dev/ttyS0

PapaSchlumpf70, 2024-01-06:
> What method are you using? The main or alt-method?
> 
> if you wipe the flash and re build the MCU Firmware in kiauh with no bootloader (instead of 32kb) then try.
> 
> stm32flash -w /home/mks/klipper/out/klipper.bin -v -g 0x8008000 /dev/ttyS0

First of all, thanks for your support.
Unfortunately, I didn't make any progress.
I tried both methods without success. The only advantage of investigating the alt method is that I don't need to touch the printer for flashing anymore :-)
The only process working for me is flashing the temp FW provided here and flashing the kiauh generated one afterward. Neither the one with 32kb Bootloader nor without generated FW survives a power cycle. Both survive a sudo reboot.
None FW did work flashing after wiping. I always have to go via the GIT-provided one.
What concerns me is the fact that even my backed-up stock FW is not working anymore.
As the printer is still working with the specific method I have the feeling(hope) that only a small piece is missing.

PapaSchlumpf70, 2024-01-06:
Unfortunately, a full wipe does not change the situation.
I followed this process(32kb boot loader enabled in the config):

PlastikSchmelzer:~:% sudo service klipper stop                                                                
[sudo] password for mks: 
PlastikSchmelzer:~:% sudo gpioset gpiochip1 15=0                                                              
sleep 0.5
sudo gpioset gpiochip1 14=1
sleep 0.5
sudo gpioset gpiochip1 15=1
sleep 0.5
sudo gpioset gpiochip1 14=0
PlastikSchmelzer:~:% stm32flash -w empty.bin -v -g 0x0 /dev/ttyS0                                             
stm32flash 0.7

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Size         : 262144
Interface serial_posix: 57600 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0423 (STM32F401xB(C))
- RAM        : Up to 64KiB  (12288b reserved by bootloader)
- Flash      : Up to 256KiB (size first sector: 1x16384)
- Option RAM : 16b
- System RAM : 30KiB
Write to memory
Erasing memory
Wrote and verified address 0x08040000 (100.00%) Done.

Starting execution at address 0x08000000... done.

PlastikSchmelzer:~:% stm32flash -w /home/mks/klipper/out/klipper.bin -v -S 0x08008000 -g 0x08000000 /dev/ttyS0
stm32flash 0.7

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Location     : 0x8008000
Size         : 32172
Interface serial_posix: 57600 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0423 (STM32F401xB(C))
- RAM        : Up to 64KiB  (12288b reserved by bootloader)
- Flash      : Up to 256KiB (size first sector: 1x16384)
- Option RAM : 16b
- System RAM : 30KiB
Write to memory
Erasing memory
Wrote and verified address 0x0800fdac (100.00%) Done.

Starting execution at address 0x08000000... done.

PlastikSchmelzer:~:% sudo service klipper start                                                               
PlastikSchmelzer:~:% sudo reboot

The system will reboot now!

What fails.

PapaSchlumpf70, 2024-01-07:
Not sure what I've done differently, but the last standard flash (stm32flash -w /home/mks/klipper/out/klipper.bin -v -S 0x08008000 -g 0x08000000 /dev/ttyS0) of the Kiauh generated FW did solve the issue. 
The FW is now working after power cycles.
Maybe it was your aura :-)
The flickering of the LED is now gone with the new FW.

Thanks a lot for your good work, the printer is working better than ever with your image.
Keep up the good work

halfmanbear, 2024-01-07:
Glad you got it sorted! Enjoy

closed 2024-01-07 by halfmanbear