OpenNeptune3D/OpenNept4une #45: Idea: Initial OpenNept4une Setup through Screen

barrenechea, 2024-01-28:
I don't know if we have enough control to achieve such a workflow, but I thought it would be nice to leverage the screen to do the initial setup.

Idea:
The display service could detect no specific printer settings on the first boot. If so, it could display a Setup page:
- It should request for the printer model (N4, N4Pro, N4Plus, N4Max)
- It should request for the printer variation (0.8, 1.2)
- It should request for the motherboard revision (v1.0, v1.1, v2.0)

Once the info is gathered, the display service could leave the screen in some loading state, trigger some bash script setting the printer with the specific .dtb, printer configs and trigger a reboot so the patches are applied.

This should be helpful mainly when flashing a fresh OpenNept4une image into the user's eMMC.

barrenechea, 2024-01-28:
@phillipthelen I would appreciate your input on this!

halfmanbear, 2024-01-28:
Yes this is a good idea, although will probably only be added once all the main features are finished with the display. 

It would make the first boot more user friendly

halfmanbear, 2024-01-28:
Could probably use the firmware reset option in the (touchscreen) settings page to re-run this setup also.

phillipthelen, 2024-01-29:
@barrenechea I have been able to figure out enough of how to communicate with the display that we could implement something like this.
We would need to load an unused page (there are still plenty left that we don't use and also plenty that are not actually implemented), clear the screen with the background grey and then draw text with backgrounds as "buttons" for the options. The display has a mode `sendxy 1` where it sends touch events via serial. This way we can check if one of the fake buttons was tapped and react to that. It's a bit messy, but would work. Along with this we could implement Wi-Fi config as well. Elegoo luckily already has screens for this that we can leverage.
As @halfmanbear said, there are probably a few things that are higher priority. Personally I would also want to have a second Emmc for testing this, since I don't necessarily want to go reset my printer a bunch while testing. 😅

I think before this can/should be implemented we'd want the following things available:
- configuring wifi from the display
- script to perform a "factory" reset
- add command line options to OpenNept4une.sh so that we can invoke it from the display controller

chuckkay, 2024-02-01:
everything is working for me, but question should I be installing KlipperScreen, will it work on elegoo monochrome screen and or will it mess up anything.

plygrnd, 2024-02-01:
Definitely install KlipperScreen - but it won't work on the built-in screen, you'll need an external display of some sort (like an old phone or a Raspberry Pi)

phillipthelen, 2024-02-01:
@chuckkay KlipperScreen will not work on the elegoo screen. They use two different methods of connecting a display. You can however install the OpenNept4une display service (you can enable that from the OpenNept4une.sh script) to get the elegoo display to work.

chuckkay, 2024-02-01:
> @chuckkay KlipperScreen will not work on the elegoo screen. They use two different methods of connecting a display. You can however install the OpenNept4une display service (you can enable that from the OpenNept4une.sh script) to get the elegoo display to work.

Yes i have the beta display service up and running, I was just wondering why klipper screen is available as an installable option in your script, is that for people using a sonic pad or similar?

DavidYummy, 2025-02-25:
Is the project dead

barrenechea, 2025-02-25:
> Is the project dead

No, please refer to https://github.com/OpenNeptune3D/OpenNept4une/discussions/247

josephgiesel, 2025-03-10:
> Yes i have the beta display service up and running, I was just wondering why klipper screen is available as an installable option in your script, is that for people using a sonic pad or similar?

Hello! I have had a lot of difficulty setting up klipper screen. Did you connect the screen via USB A or C?