OpenNeptune3D/OpenNept4une #46: MCU Protocol error

SrPhilippe, 2024-01-28:
I just went through the step-by-step installation and after finishing I got this error.

I have a NP4, V1.0 PCB version.

[image]

SrPhilippe, 2024-01-28:
I also followed this step and after updating the MCU the file X_4.bin changed it's name to X_4.cur which means the update was successful.
[image]

halfmanbear, 2024-01-28:
Your error is because the MCU (STM32) is showing as v0.10.0 still and not v0.12.0.
Its odd the file renamed to CUR yet didn't seem to flash the latest version.

Have you power cycled the machine since? Might be worth re-generating the .bin files and trying again to flash the MCU

SrPhilippe, 2024-01-28:
Ok. Now my MCU is running V0.12.0 and i'm still getting the error
could this be related to RPI version, how do I update it?
[image]

halfmanbear, 2024-01-28:
Ok yes you need to update your Virtual MCU RPi re-run the script and select (Virtual RPi) when asked

SrPhilippe, 2024-01-28:
> Ok yes you need to update your Virtual MCU RPi re-run the script and select (Virtual RPi) when asked

It worked out!
I was checking all MCUs update before (option 4). Now I did Virtual RPi (option 2). The error message is gone!

closed 2024-01-28 by SrPhilippe