OpenNeptune3D/OpenNept4une #51: Allow updating through the touch screen

phillipthelen, 2024-01-31:
Bring back the update button on the About screen and make it run `opennept4une update_repository`. Ideally should show a progress indicator and prompt for confirmation if there is an update and ask if the user wants to apply it.

halfmanbear, 2024-01-31:
I've got venv and pycache in gitnore so as long as the screen installer doesn't need running to get new pips etc it should be fine. But it might break the screen if it does need new packages etc

barrenechea, 2024-01-31:
Could the screen query data from Moonraker's Update Manager? Similar to what KlipperScreen does to list all pending updates.

OpenNept4une should be listed as an updatable package on the next release, so listing all Update Manager packages should cover the issue and allow updating Klipper, OS packages, etc.

halfmanbear, 2024-01-31:
It should already show up in the fluidd / moonraker update manager. It was added last week

barrenechea, 2024-01-31:
> opennept4une update_repository

Sure, what I meant was for the display to support Update Manager and allow for updating all things handled by it, instead of just running `opennept4une update_repository`