OpenNeptune3D/OpenNept4une #56: Feature Request Touchscreen - Probe Screws individually

DocsyDynamo, 2024-02-01:
As for now we use the screw_tilt_calculate feature to manually level the bed. 
So you start the macro, wait until all 4 screws are probed and adjust the leveleing knobs accordingly to the resullt.
But for me its quite hard to know what "20 Minutes" are for example. So you are guessing somehow how much to turn. Then you restart the macro and do it all over again until the result is good enough for you. Thats quite time consuming. 

So I would love a feature that allows you to probe  each screw individually via the touchscreen and you get back the probing results in realtime. So first you probe your reference screw (e.g screw_1). Lets say the result is 1.53. Next you probe screw_2. Whilte getting the probing result on screen you can turn the leveling knobs until the result is 1.53 as well. Next you probe scew_3 and so on. 

Manual leveling the bed by probing each screw/corner individually would be much faster and more precise. I made a sketch to get the idea. 

This guy has this in his tool "bed leveler 5000" (skip to minute 5:30 in the video):
https://www.youtube.com/watch?v=JBGN3U0C2LM&ab_channel=S%26MMakers

[image]

DocsyDynamo, 2024-02-06:
I've created  Macros for moving the print head to the screw positions + probing. (N4+N4Pro coordinates). 
Tapping one of the buttons should execute  the corresponding macro. 

[gcode_macro probe_at_rear_left]
gcode:
    G1 Z10 F500 
    G1 X56.75 Y182.05 F5000 
    PROBE PROBE_SPEED=10 SAMPLES=5 SAMPLE_RETRACT_DIST=3.0 SAMPLES_TOLERANCE=0.01 SAMPLES_TOLERANCE_RETRIES=20 SAMPLES_RESULT=median
    G1 Z10 F500 

[gcode_macro probe_at_rear_right]
gcode:
    G1 Z10 F500 
    G1 X226.75 Y182.05 F5000 
    PROBE PROBE_SPEED=10 SAMPLES=5 SAMPLE_RETRACT_DIST=3.0 SAMPLES_TOLERANCE=0.01 SAMPLES_TOLERANCE_RETRIES=20 SAMPLES_RESULT=median
    G1 Z10 F500 

[gcode_macro probe_at_front_right]
gcode:
    G1 Z10 F500 
    G1 X226.75 Y12.05 F5000
    PROBE PROBE_SPEED=10 SAMPLES=5 SAMPLE_RETRACT_DIST=3.0 SAMPLES_TOLERANCE=0.01 SAMPLES_TOLERANCE_RETRIES=20 SAMPLES_RESULT=median
    G1 Z10 F500 

[gcode_macro probe_at_front_left]
gcode:
    G1 Z10 F500 
    G1 X56.75 Y12.05 F5000 
    PROBE PROBE_SPEED=10 SAMPLES=5 SAMPLE_RETRACT_DIST=3.0 SAMPLES_TOLERANCE=0.01 SAMPLES_TOLERANCE_RETRIES=20 SAMPLES_RESULT=median
    G1 Z10 F500

clmcgrath, 2024-10-18:
if you add the [bed_screws] section with the same values from tilr adjust you can enable stock like auxillary levelling , i agree it would be nice to have both of these in the ui but for now you can use this to set your screws manually then use tilt adjust to fine tune