OpenNeptune3D/OpenNept4une #69: New Macros UNLOAD & LOAD_FILAMENT

halfmanbear, 2024-02-02:
Create and add the following macros to all printer.cfg

UNLOAD_FILAMENT
LOAD_FILAMENT

halfmanbear, 2024-02-05:
Added to dev branch

closed 2024-02-05 by halfmanbear

fifonik, 2024-02-12:
These macroses did not work for me (N4) with message 'Extrude below minimum temp' until I added 'min_extrude_temp' in [extruder] section (there was 'min_temp' there)

halfmanbear, 2024-02-13:
Don't remove min_temp, if no min_extrude_temp is specified here it defaults to 170°C

The error is there for a reason you shouldn't be loading or unloading the filament below 170°C

https://www.klipper3d.org/Config_Reference.html?h=min_temp#extruder

fifonik, 2024-02-13:
When I tested, I heated up extruder to 170 and load/unload did not work from the screen. 
I removed my min_extrude_temp from config, restarted, heated extruder to 175 (from web UI) and all good. Probably the temperature was a bit below the minimum limit.
No issue then.
Thanks.