OpenNeptune3D/OpenNept4une #70: Wiki page with board Information

Novus4K, 2024-02-02:
Creat a new Wiki page with information about the different board revisions. (Source: official Elegoo manual)

[image]
[image]
[image]
[image]

phillipthelen, 2024-02-02:
Was the first image uploaded twice accidentally? I can't actually tell the difference. I'll add a page for this!

Novus4K, 2024-02-02:
That was an accident. I have removed it and sorted the others by version.