OpenNeptune3D/OpenNept4une #72: HOST overheating when not printing

kwe1009, 2024-02-02:
I get the following error after my 4 Pro sits for 10-20 minutes powered on but not printing.  The only thing I can do is turn it off and wait for it to cool down.  This never happened before on stock firmware.  Any suggestions?

HOST temperature 83.5 above maximum temperature of 80.0.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

halfmanbear, 2024-02-02:
Is your main board fan under the printer spinning at all. If you move an axis or turn on a heater it should turn on (will stay on for 15 minutes after Amy heating or movement stops).

Many people have found hot glue has dropped onto the fan there and stopped it spinning.

kwe1009, 2024-02-09:
I checked the main board fan and it was OK.  Oddly enough I haven't had the problem since.

closed 2024-02-09 by kwe1009