OpenNeptune3D/OpenNept4une #73: pause on runout...

chuckkay, 2024-02-03:
I dont understand something, the Pause on filament runout is working fine, but i was just looking at printer.cfg file and it is set to false, how is that possible it's set to false but it's still doing it's job..

#####################################################################
# Filament sensor
#####################################################################

[filament_switch_sensor fila]
pause_on_runout: False
runout_gcode:
            M600
insert_gcode:
event_delay: 3.0
pause_delay: 1.0
switch_pin: PA12

halfmanbear, 2024-02-03:
Because I don't want it to preform a normal PAUSE action straight away. I've got it to trigger my M600 instead. The M600 has the pause function call within it so ultimately it does pause even though it says false.

chuckkay, 2024-02-03:
ok thanks,
maybe a #comment line "do not touch", 🤓as people gonna monkey around.

closed 2024-02-03 by chuckkay