OpenNeptune3D/OpenNept4une #74: front USB-C port - can it be enabled and be used

chuckkay, 2024-02-03:
i read somewhere at the front USB-C port is for internal use by the manufacturer and not usable as standard USB, was wondering if the port could be enabled and used,  and if it provides enough power  for a big tree tech LCD screen for Klipper screen

plygrnd, 2024-02-03:
[image]

In the above picture, the USB-C port (labeled HUS) is wired directly into the [CH340C](https://wch-ic.com/products/CH340.html) USB bus converter above it ([source](https://r6e.github.io/neptune_4/hardware.html))

closed 2024-02-04 by chuckkay

halfmanbear, 2024-02-04:
Yes its for the serial terminal connection. It is enabled and you can use it for this purpose.