OpenNeptune3D/OpenNept4une #85: Display stops updating when probing bed before print start

scherlund, 2024-02-08:
When a print is started the screen stops updating when adaptive meshing starts.

The screen doesn't freeze as the little animations for the bed temp and Fan still moves, but no new information is shown.

This is using the latest OpenNept4une and Display connector (Stable) is of 10 minutes ago. Both updated through the script.

Klipper and everything else is updated through KIAUH.

It is possible to manipulate other thing through the screen while printing like the LEDs. However the button doesn't change. But that could be a separate issue (It's BETA after all)

Attached is a picture of a print in progress (Temp Tower).
[image]

phillipthelen, 2024-02-08:
Fixed in both dev and main branch. Updating through Fluidd or OpenNept4une.sh should fix everything :)

closed 2024-02-08 by phillipthelen

scherlund, 2024-02-08:
Can confirm. With an update it works. It even shows the probing progress now :)

You guys rock.

Justdigit, 2024-02-08:
Off topic question.... Why you are updating via Kiauh?? And not within fluidd?? Is. That different? Or beteer?

scherlund, 2024-02-08:
> Off topic question.... Why you are updating via Kiauh?? And not within fluidd?? Is. That different? Or beteer?

Neither worse nor better :) 

It's just a personal preference. I like to SSH when doing updates and stuff like that. I use Fluidd/Mainsail for normal printing :)

Justdigit, 2024-02-08:
> > Off topic question.... Why you are updating via Kiauh?? And not within fluidd?? Is. That different? Or beteer?
> 
> Neither worse nor better :)
> 
> It's just a personal preference. I like to SSH when doing updates and stuff like that. I use Fluidd/Mainsail for normal printing :)

OK thanks for the quick answer.