OpenNeptune3D/OpenNept4une #101: Support running Input shaper from display

phillipthelen, 2024-02-12:
At least on Plus and Max models we can offer input shaper from the display. I still need to look into how exactly the Flow and UX for that is.
Could also try to detect if the normal and pro model have the definitions for input shaper in the printer config to then also offer it there. But that's a bit more involved.

barrenechea, 2024-02-13:
I just found some takes on Input Shaper for the Max, hope it helps! https://www.youtube.com/watch?v=7DWOqmXAABY&t=115s

phillipthelen, 2024-02-13:
awesome! That luckily doesn't look too hard

Justdigit, 2024-02-13:
But does the Pro has the build in accelerometer in the print head I mean.

halfmanbear, 2024-02-13:
Only the Max and the Plus i believe

MuteKnee3D, 2025-04-03:
Just out of curiosity (and slightly off topic) has anyone tried mounting a plus/max toolhead board (with onboard axdl345) to a N4/Pro toolhead? I suppose you would need an additional ribbon cable with the proper JST connectors but that shouldn't be too tough. I only have a Plus so no experience with the mount or the v1 mobo on the others. The toolhead boards for the plux/max are relatively cheap.