OpenNeptune3D/OpenNept4une #106: Screws tilt adjustments does not display values & hints when they in red zone

fifonik, 2024-02-12:
Installed it on Neptune 4.
Started the screw tuning and after probing it does not show any hints on screen.
At the beginning I thought everything is good as the bed is aligned (it was aligned before flashing OpenNeptune3D):
[image]

Print failed and later I noticed that in web UI it actually shows a popup with red values. Started to tune it up and when reach "green zone" it is displayed on screen.
[image]

So for now it is impossible for me to adjust screws without web UI.

Justdigit, 2024-02-12:
Do yourself a favor... Get 4 silicone spacers instead of the springs.
Search on YouTube bed leveler 5000.
And be done with it.....

fifonik, 2024-02-12:
Will it fix the screen's UI? :-P

closed 2024-02-18 by phillipthelen