OpenNeptune3D/OpenNept4une #108: LAN connectivity issue

fifonik, 2024-02-13:
I have my printer (N4) in garage and it is connected to home network through LAN port (LAN <=> WiFi extender <=> Main WiFi router).
When I want to print I'm turning on the extender and printer simultaneously with one wall socket switch.
However, the printer with OpenNeptune is loading faster than the WiFi extender and as a result -- it does not have IP (from DHCP).
If I unplug LAN cable from printer and plug it in again it still do not get IP.
As a result, now I have to turn ON extender first, wait till it is loaded and only then turn ON the printer.

With original Elegoo firmware everything works with no issues -- I turning ON both devices simultaneously and after a while I can access the printer over LAN.

Can linux networking be adjusted for such situation?

Thanks.

halfmanbear, 2024-02-13:
You could try running,

```
sudo nmcli con mod eth0 connection.autoconnect yes
```
Then
```
sudo reboot
```

fifonik, 2024-02-15:
Issued the commands, but no luck:
Then turned everything off (printer + WiFi extender), waited, turned on, waited -- cannot connect to printer and no IP on the screen.
Unplugged LAN cable, waited ~10 secs, plugged it again, waited ~10 secs -- cannot connect to printer and no IP on the screen.

halfmanbear, 2024-02-16:
You can connect via serial to the usb-C port to run the following to see what's happening.

```
ip a
```

Can try reconfigure the connection with.

```
sudo nmtui 
```

closed 2024-02-28 by halfmanbear