OpenNeptune3D/OpenNept4une #109: TMC 'stepper_y Overtemp error

Robotter89, 2024-02-13:
Hello,

I installed to openNept4une source today on my 4 Max, everything went well until i was starting a test print.
About 45% in the print the printer stopped and turned off.
The error message in fluidd web interface is : TMC 'stepper_y' reports error: DRV_STATUS: 001901c3 otpw=1(OvertempWarning!) ot=1(OvertempError!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) t120=1 cs_actual=25

How to fix this?

I see that the stealthchop_threshold, run and hold current on stepper motor Y is a bit different from the printer.cfg I had with Elegoo's firmware.

halfmanbear, 2024-02-13:
What values did you have in elegoos configs the driver / stepper sections?

Robotter89, 2024-02-13:
> What values did you have in elegoos configs the driver / stepper sections?

Here is my lates values from elegoo:
[tmc2209 stepper_x]
uart_pin: PB9
run_current: 1.0
hold_current: 0.8
interpolate: True
stealthchop_threshold:0
driver_SGTHRS:90
diag_pin:^PC0

[tmc2209 stepper_y]
uart_pin: PD2
run_current: 1.4
hold_current: 1.0
interpolate: True
stealthchop_threshold:0
driver_SGTHRS:80
diag_pin:^PB8

[tmc2209 stepper_z]
uart_pin: PC5
run_current: 0.8
interpolate: True
stealthchop_threshold: 0

Robotter89, 2024-02-13:
I have tried to set the same values as the elegoo config, but still the same problem.

halfmanbear, 2024-02-13:
Is your mainboard fan spinning ok? 
It should spin up when the steppers are activated without any heating, for example after homing the printer.

Robotter89, 2024-02-13:
It could look like it was due to i haven't mounted the cover under the printer when doing the test print after the installation, so it was for sure a cooling issue on the mainboard.  Now the printer is working fine with the cover back in place and the fan running, showing good temps on MCU and Host.

Thank you both.

closed 2024-02-13 by Robotter89