OpenNeptune3D/OpenNept4une #117: Kernel Panic - Neptune4+Pro-v0.1.5-ZNP-K1-V1.0.img - cannot boot after installing image to emmc

Tylerjet, 2024-02-19:
Running an N4Pro With ZNP K1 V1.0 and installed the `Neptune4+Pro-v0.1.5-ZNP-K1-V1.0.img`.

I get the following output from serial over USB:
```
DDR version 1.16 20190528
ID:0x805 N
In
DDR3
333MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
ddrconfig:1
OUT
Boot1 Release Time: May 13 2019 17:34:36, version: 2.50
ChipType = 0x11, 233
mmc2:cmd19,100
SdmmcInit=2 0
BootCapSize=2000
UserCapSize=7456MB
FwPartOffset=2000 , 2000
SdmmcInit=0 NOT PRESENT
StorageInit ok = 11429
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT 0x337a9f0 signature is wrong
recovery gpt...
GPT 0x337a9f0 signature is wrong
recovery gpt fail!
LoadTrust Addr:0x4000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xb78e0
RunBL31 0x40000
INFO:    Preloader serial: 2
NOTICE:  BL31: v1.3(debug):403e0b816
NOTICE:  BL31: Built : 14:13:08, Aug 11 2020
NOTICE:  BL31:Rockchip release version: v1.3
INFO:    ARM GICv2 driver initialized
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2022.07-armbian (Dec 13 2023 - 17:09:44 +0000)

Model: Makerbase mkspi
DRAM:  1022 MiB
PMIC:  RK8050 (on=0x10, off=0x00)
Core:  228 devices, 23 uclasses, devicetree: separate
MMC:   mmc@ff500000: 1, mmc@ff520000: 0
Loading Environment from MMC... Card did not respond to voltage select! : -110
*** Warning - No block device, using default environment

In:    serial@ff130000
Out:   serial@ff130000
Err:   serial@ff130000
Model: Makerbase mkspi
Net:   No ethernet found.

Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
3548 bytes read in 3 ms (1.1 MiB/s)
## Executing script at 00500000
Boot script loaded from mmc 0:1
190 bytes read in 2 ms (92.8 KiB/s)
21622691 bytes read in 490 ms (42.1 MiB/s)
29729280 bytes read in 672 ms (42.2 MiB/s)
64614 bytes read in 7 ms (8.8 MiB/s)
2698 bytes read in 5 ms (526.4 KiB/s)
Applying kernel provided DT fixup script (rockchip-fixup.scr)
## Executing script at 09000000
Unknown command 'kaslrseed' - try 'help'
Moving Image from 0x2080000 to 0x2200000, end=3f00000
## Loading init Ramdisk from Legacy Image at 06000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    21622627 Bytes = 20.6 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Ramdisk to 3ca80000, end 3df1ef63 ... OK
   Loading Device Tree to 000000003ca07000, end 000000003ca7ffff ... OK

Starting kernel ...

efi_free_pool: illegal free 0x000000003cf21040
efi_free_pool: illegal free 0x000000003cf1e040
efi_free_pool: illegal free 0x000000003cf1c040

done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Will now check root file system ... fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p2] fsck.ext4 -a -C0 /dev/mmcblk1p2
armbi_root: clean, 100666/434720 files, 1117369/1798144 blocks
done.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.
[    5.700904] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
/sbin/init: error while loading shared libraries: /usr/lib/aarch[    5.701621] CPU: 3 PID: 1 Comm: init Not tainted 6.1.67-current-rockchip64 #2
[    5.702729] Hardware name: Makerbase mks-pi (DT)d ELF header

[    5.703645] Call trace:
[    5.703877]  dump_backtrace+0xd8/0x130
[    5.704236]  show_stack+0x18/0x30
[    5.704551]  dump_stack_lvl+0x68/0x84
[    5.704903]  dump_stack+0x18/0x34
[    5.705221]  panic+0x184/0x35c
[    5.705513]  do_exit+0x9b8/0x9c0
[    5.705820]  do_group_exit+0x34/0x90
[    5.706157]  __arm64_sys_exit_group+0x18/0x20
[    5.706564]  invoke_syscall+0x48/0x114
[    5.706926]  el0_svc_common.constprop.0+0x44/0xfc
[    5.707371]  do_el0_svc+0x30/0xd0
[    5.707691]  el0_svc+0x2c/0x84
[    5.707989]  el0t_64_sync_handler+0xbc/0x140
[    5.708395]  el0t_64_sync+0x18c/0x190
[    5.708745] SMP: stopping secondary CPUs
[    5.709175] Kernel Offset: disabled
[    5.709498] CPU features: 0x00000,00800080,0000421b
[    5.709948] Memory Limit: none
[    5.710241] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00 ]---
```

Was able to boot into armbian with the base image V0.1.6 from the [Armbian-ZNP-K1-build](https://github.com/OpenNeptune3D/Armbian-ZNP-K1-build/releases/tag/v0.1.6) repo.

Currently installing V0.1.4 to the emmc to see if that one is fine as i believe that is the one i was running previously without issue.

Tylerjet, 2024-02-19:
Confirmed V0.1.4 can be booted into just fine.

Tylerjet, 2024-02-20:
Yeah I notice after installing v0.1.6 that it was just the base image with nothing klipper related  installed or a user account setup. So that is why I also tried v0.1.4.

It definitely is odd though is there any files I should maybe try and pull from the older version to try and resolve this or maybe try rebuilding from source on my end just to see the result?

halfmanbear, 2024-02-20:
The error is regarding,
/sbin/init, (a symbolic link to)
/lib/systemd/systemd 

I'm not sure what to suggest, might be worth zeroing out the eMMC with dd then downloading v0.1.5 again?

Are you flashing them with etcher or dd?

Tylerjet, 2024-02-20:
Hrmm alright can try that would deleting all the partitions on the emmc and then reformatting also work (running on windows for main pc) and tried once with usbimager and once with etcher with the same result on both.

Tylerjet, 2024-02-20:
Seems to be working now! Just to be safe i deleted all the partitions but then also did a write 0 wipe on the emmc and then redownloaded and burned the image with USBImager and this time booted up just fine.

closed 2024-02-20 by Tylerjet

halfmanbear, 2024-02-20:
Awesome glad you got it sorted!

Tylerjet, 2024-02-21:
One final side note/question @halfmanbear, after doing a git pull of the OpenNept4une folder to update it to latest and running the script the Model section of the header is blank? does that not get update until installing/updating the printer configurations?

As Im pulling back in my backed up configurations but still want to make sure everything for that repo is setup properly when running any updates for other items etc...

Tylerjet, 2024-02-21:
Ahh nvm answered my own question, so instead i will change it to would you be opposed to a PR that adds another advanced option that presents just that part of the menu to sets the model information without having to go through the entire printer config section just to set those?

halfmanbear, 2024-02-21:
You have to go through the printer config section to get those, because the script is generic to all printers models. 

As the printer config install is required during the setup I don't see this as an issue. 

After running this a flag is created in /boot/.OpenNept4une detailing what model Printer you selected. The display service uses this also.

Tylerjet, 2024-02-21:
ahh ok I'll just be sure to add that as something to backup then.