OpenNeptune3D/OpenNept4une #124: screw tilt adjust on display obstructed

chuckkay, 2024-02-26:
the retry button obscures the values and ccw cw on the 2 middle screw measurements...  on right can't tell if it says ccw or cw, on the left i cant see the last digit

phillipthelen, 2024-03-10:
In newer versions of the macro we disabled bed heating for this. It’s not necessary and makes it easier to run the screw adjustment.
Green means the screw is in the acceptable/good range. They don’t all need to be at exactly 0 according to Klipper docs. Below 5 “minutes”, they are already good enough.

the other things are bugs I’ll look into

clmcgrath, 2024-10-18:
temporary workaround : open fluidd and run tilt adjust either from fluidd ui or from the touch screen and you will get a popup ui that gives the values

closed 2025-03-10 by halfmanbear