OpenNeptune3D/OpenNept4une #132: filament run out code flawed

chuckkay, 2024-02-29:
sensor working fine on runout, the print pauses and toolhead parks 
but then shuts down the extruder heater with M104 S0   

but when the you reload filament and hit resume nothing happens... this is because of 
of this code in the resume macro
{% if etemp > 0 %}
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={etemp|int - 4} MAXIMUM={etemp|int + 10}  ; wait for hotend to heat back up
{% endif %}
This section waits for the hotend to heat back up to its previous temperature before resuming the print. However, if the hotend was turned off during the pause (as mentioned above), the nothing will happen..

so instead of M104 S0  i changed it to M104 S160.... 

and then in the resume code i modified this 
{% if etemp > 0 %}
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={etemp|int - 4} MAXIMUM={etemp|int + 10}  ; Wait for hotend to reach near target temperature
{% endif %}
  
to this
{% if etemp > 0 %}
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET={etemp|int}  ; Heat the hotend back up to the original temperature
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={etemp|int - 4} MAXIMUM={etemp|int + 10}  ; Wait for hotend to reach near target temperature
{% endif %}

both are temporary fixes until the problem is addressed through an update..

I hope i'm right and not the only one experiencing this problem,  also i couldn't find a reason but it also seemed that the bed was cooling down also, but i couldn't find any code that was causing that, so just to be safe i added  M140 S{printer.heater_bed.target}   to the Pause macro...

Also i get 2 idle_timeout messages i assume the 7200 is to give me 2 hours to change the filament and once the filament is changed it reverts back to 600 for protection for any other idle reasons

halfmanbear, 2024-02-29:
https://github.com/OpenNeptune3D/OpenNept4une/issues/133#issue-2160853033

halfmanbear, 2024-02-29:
Fixed in `dev`, please re-open if you find any issues
https://github.com/OpenNeptune3D/OpenNept4une/commit/adb3b49f159ee23474cf2ac57208ecf1f2d4e538