OpenNeptune3D/OpenNept4une #136: mainsail.cfg missing error

antoncl, 2024-03-01:
If mainsail has been uninstalled then an error occurs during startup with the message: Unable to open config file /home/mks/printer_data/config/mainsail.cfg

out commenting the [include mainsail.cfg} in printer.cfg solves the problem, but then the update manager in moonraker balks. Which again can be remedied by outcommenting the relevant update section.

halfmanbear, 2024-03-01:
If you were to uninstall anything from the standard image you would have to remove the relevant section from the include section of `printer.cfg` and `moonraker.conf`.

So what's the issue, after doing both of the above?

antoncl, 2024-03-02:
There isn't any big issue here. Removing the sections (out commenting) removes the error message. The only issue I see is the 'user_settings.cfg' which, in my mind, tells me to restrict any user modifications to that file. (I'd expect modifications I make to any other file to disappear willy nilly)

So... it is more of a the-end-user-may-be-confused (or I-am-confused-by-the-meaning-of-the-name) issue than anything else.

halfmanbear, 2024-03-02:
Yes, the recommended practice is to restrict any user modifications to that file.

https://github.com/OpenNeptune3D/OpenNept4une/wiki/Updating-your-printer.cfg

As explained in the wiki, if you were to update your `printer.cfg` with the script down the line. Your previous `printer.cfg` is backed up (with any changes you have made). 

So nothing will disappear, this new feature doesn't remove anything that was there before. It just offers a way for users to get the latest macros (from us) without having to copy and paste their edits from the backup to the new `.cfg`.

closed 2024-03-02 by halfmanbear