OpenNeptune3D/OpenNept4une #138: Pinout information for E1 Axis stepper

TheMeanCanEHdian, 2024-03-08:
Just found this project today and as someone who went through the effort to get my Neptune 4 Pro on standard software I love to see it!

I am not sure how this might fight into the project, but I wanted to share the E1 Axis pinout information that I am using for a dual z axis setup. (in case you didn't already have this info). I have a v1.0 board, I am not sure if this is different for the other versions.

```
step_pin:PB1
dir_pin:PB2
enable_pin: !PB0
```

Thanks for the work you are doing here, looking forward to the developments.

halfmanbear, 2024-03-08:
I've not done the mod but I made the config below for the v1.0 n4 / pro.
Is my `uart_pin: PB6` correct?

```
[z_tilt]
z_positions:
  -30, 117.5
  265, 117.5
points:
  30, 97.05
  235, 97.05
speed: 50
horizontal_move_z: 5
retries: 10
retry_tolerance:  0.003

[stepper_z1]
step_pin: PB1
dir_pin: PB2
enable_pin: !PB0
microsteps: 32
rotation_distance: 8

[tmc2209 stepper_z1]
uart_pin: PB6
run_current: 0.5
interpolate: false
```

TheMeanCanEHdian, 2024-03-08:
Oops, sorry for not sharing that haha. Yes PB6 is correct.

TheMeanCanEHdian, 2024-03-08:
Can you help me understand how you chose the z_tilt positions and points you used? Mine is a little different, but I was also just doing my best approximation based on the Klipper documentation (no in depth measurements or anything). For the record this is what I have:
```
z_positions:
    -30.66, 0
    266.34, 0
points:
    40,115
    235,115
```

barrenechea, 2024-03-10:
I can provide some info since I did the mod quite a while ago:

`z_positions` refers to the physical location of the center of each motor in (x, y) coordinates from the bed, being `y` irrelevant from my understanding since it is a bedslinger. The first line, `-30.66, 0`, refers to the position of the Z-stepper, and the second line, `266.34, 0`, refers to the physical location of the Z1-stepper. Depending on how you wired Z and Z1, these may need to be switched around. I measured -30cm and +30cm for each side, but yours may be a bit more precise (if the measurement is accurate).

`points` refers to the locations on the bed where the probing will be made, to adjust the Z-tilt. The ordering should match the `z_positions` ordering for Z and Z1, depending on how the steppers were wired.

My recommendation is to probe the same points used for the bed leveling since they are positioned where the screws are (and we can adjust them).

This is what I have set:
```
[z_tilt]
z_positions: 
  -30, 0
  265, 0
points: 
  56.75, 182.05 ; rear left screw
  226.75, 182.05 ; rear right screw
speed: 100
horizontal_move_z: 5
retries: 10
retry_tolerance: .0025
```

I hope it helps!

TheMeanCanEHdian, 2024-03-11:
Good call about using the screw positions for the probe points. I did follow the Klipper docs to try and get the positions as accurate as I could, I just didn't have any kind of model to ensure the measurements 100%.

Firefighter1023, 2024-05-26:
Would the pinouts listed above also work for the 4 max/plus?

Rickwo15, 2024-06-10:
why the Z axis have different setting microstepps in comparing with elegoo printer.cfg?
In original setting is microstepps on 16stepps but the stepper physically is 42-32stepper motor.
In openneptune printer.cfg is value 32 which seems to be correct but maby elegoo have some point why they put half value.
The question is why you using the value 32 instead 16 as original.
[stepper_y]
step_pin:PB4
dir_pin:PB3
enable_pin:!PB5
microsteps:16
rotation_distance: 40
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin:PB8
position_min: -2
position_endstop:0 
position_max:235
homing_speed:50
homing_retract_dist:5
homing_positive_dir:false