OpenNeptune3D/OpenNept4une #146: Cannot add config to Orca

rashevskyv, 2024-03-13:
Using version OrcaSlicer V2.0.0 Beta Release

File -> Import -> Import configs -> Elegoo Neptune 4 (0.4 nozzle)-OpenNept4une.orca_printer

[image]

I have already 4 printers on orca's config, include vanilla N4. Please help me add OpenNept4une's config

halfmanbear, 2024-03-13:
Have you already set up orcas default profiles for the N4 first as per

https://github.com/OpenNeptune3D/OpenNept4une/wiki#importing-orcaslicer-profiles

Does this work in the non beta latest release orcaalicer version as suggested?

Kemar0611, 2024-03-31:
I did the same thing. 
[image]
[image]

closed 2024-04-16 by halfmanbear