OpenNeptune3D/OpenNept4une #169: Direct connection via LAN, no IP

writepatrick, 2024-04-05:
Hi,
I have a similar but different problem as in #108 discribed:

My use case before flashing OpenNept4une was to directly connect my laptop via Ethernet with the Neptune 4 pro. I would get an IP and could connect to it directly. The printer is located somewhere, where no network access is possible.

Since the upgrade to OpenNept4une I can only access the printer if it is connected to a router, since the IP will not be assigned even after reboot when connected to the laptop.

Any suggestions how to solve this?

halfmanbear, 2024-04-05:
I'm surprised a direct connection to the laptop ever worked unless you specifically configured the laptop as a DHCP server / router. 

The other option is to use a usb c to usb a cable and connect to the printers usbc serial port via putty then set up WiFi on the printer to continue the setup.

writepatrick, 2024-04-05:
The initial setup is already completed, I brought the printer temporarily to the network to make it work and configure it. 'But now I'd like to put it back to it's place.

on Stock firmware, I did not change anything in my config for a direct connection. I just gave it a try and it worked. 

I am wondering if going the wifi route is working by opening an access point on my laptop on which the printer will connect then.

halfmanbear, 2024-04-05:
Printer WiFi connected to the router, and laptop connected to the same router over WiFi or Eth will do it.

writepatrick, 2024-04-05:
Hi this would mean I would need a explicit router for this setup. What a bummer. 

I will try to do it via AP from the laptop, maybe it will work.

halfmanbear, 2024-04-05:
No you just need to connect the printer by WiFi so the same local network as your laptop either by WiFi or ethernet that's all

writepatrick, 2024-04-05:
@halfmanbear I think we are talking past each other. there is no local network present at the location printer will be permanently installed, but I'd like to access fluidd via Orca Slicer or mainsail.

Nevermind. If I find a solution to get direct connection without the need of a network (router) I will post it here.

closed 2024-04-05 by writepatrick

halfmanbear, 2024-04-05:
Ahh I understand now. No network, you can configure your laptop like this.

https://alexanderhoughton.co.uk/blog/connecting-to-raspberry-pi-direct-to-pc-via-ethernet/