OpenNeptune3D/OpenNept4une #171: PID_tune _outer_BED saves duplicate to Save_Config section

dcchillin46, 2024-04-06:
after running PID tuning for outer bed heater, it saves under [heater_generic heater_bed_outer]. this creates a new section and leaves [heater_bed_outer] section at the top of save config section of printer.cfg.

halfmanbear, 2024-04-06:
Yes it's because I found a big in klipper and they have only just fixed it recently. PID tuning the outer bed never did anything under the old section heading. They have fixed it although it will just add a new section in save_config.

You can delete the section that's missing heater_generic.

dcchillin46, 2024-04-06:
fair enough, wasnt sure if it was an oversight. i ended up deleting the other section, just thought id let ya know.

thanks

closed 2024-04-16 by halfmanbear