OpenNeptune3D/OpenNept4une #178: Starting a print from an existing GCODE file after a print finishes causes the head to crash into the bed (it loses homing).

namalahideaway, 2024-05-02:
1. Start a print
2. Let print finish.
3. remove print from bed.
4. Click the same gcode and hit print (to print it a second time.)
5. Head doesnt home properly and crashes into bed.

klipper.log attached.
[klippy.log](https://github.com/OpenNeptune3D/OpenNept4une/files/15184823/klippy.log)

phillipthelen, 2024-05-02:
Are you starting the print from fluidd/mainsail?
Has this happened with different gcode files for you?

halfmanbear, 2024-05-02:
Also are you on a plus or max machine? (Sensorless homeing)

namalahideaway, 2024-05-04:
On a max. The homing is all wonky now. it wont reliably home. it keeps forgetting where it is.... and the y axis just forgets and doesnt move.

halfmanbear, 2024-05-04:
Are you on Dev? There are some new homing improvements

namalahideaway, 2024-05-05:
im not on dev. how do i get on dev.

The homing isnt working at all. about 90% of the time it loses where the y axis homing is and trys to print off the top (backside) of the bed.

ringnutz, 2024-05-05:
I'm on the max and not seeing this behavior at all

halfmanbear, 2024-05-05:
> im not on dev. how do i get on dev.
> 
> The homing isnt working at all. about 90% of the time it loses where the y axis homing is and trys to print off the top (backside) of the bed.

Run `opennept4une` go to the advanced page then select switch branch option. Then go back and select option 1 to install the latest printer.cfg

halfmanbear, 2024-05-07:
Those changes have moved to main, make sure you update everything in fluidd then install the latest printer.cfg via running `opennept4une` in the terminal

closed 2024-07-16 by halfmanbear