OpenNeptune3D/OpenNept4une #194: At end of print noozle starts going down smashing into the print

Sulginger, 2024-06-26:
Hey, at the end of my prints since switching to OpenNept4une the noozle starts going down straight into the print destorying it for some season.
I tried adding G1 Z10 into the "PRINT_END" command but it changes nothing, anyone got a solution for me please? Its kinda frustrating  :(

halfmanbear, 2024-06-26:
Can you upload your printer.cfg and screenshot the machine gcode from the slicer profile your using

halfmanbear, 2024-06-26:
Hmm there is nothing there with a downward z move. Anything show in the klipper console?

Sulginger, 2024-06-26:
No not rly anything sus

Sulginger, 2024-06-26:
Do you have any PRINT_END gCode which works? I'd just copy that and paste it into my printer.cfg

lexxxel, 2024-07-03:
> Ficked it by updating and installing the whole thing new

Can this issue then be closed?

closed 2024-07-03 by Sulginger