OpenNeptune3D/OpenNept4une #197: Issue with Z Offset

Choccy-vr, 2024-07-09:
Hello I recently came across a issue with the z offset page. It seemingly does not change the z offset nor display it. Also it would be nice if it automatically homes and goes to Z 0 when you click the z probe offset

halfmanbear, 2024-07-16:
Z offset page on the touchscreen? If so could you place your issue in that repository please. https://github.com/OpenNeptune3D/display_connector

If not there is some more z_offset info below.

https://github.com/OpenNeptune3D/OpenNept4une/wiki/Printer-Calibration-%E2%80%90-Klipper-&-OrcaSlicer#more-info-on-z_offset

DoomerDGR8, 2024-07-27:
> Z offset page on the touchscreen? If so could you place your issue in that repository please. https://github.com/OpenNeptune3D/display_connector

I didn't realized this... is https://github.com/OpenNeptune3D/display_connector already part of the main OpenNept4une upgrade or it is an optional addon?

halfmanbear, 2024-07-27:
> > Z offset page on the touchscreen? If so could you place your issue in that repository please. https://github.com/OpenNeptune3D/display_connector
> 
> I didn't realized this... is https://github.com/OpenNeptune3D/display_connector already part of the main OpenNept4une upgrade or it is an optional addon?

That is the repo for the display code your currently running (part of main Opennept4une). We just keep it sperate to make things tidier / easier. It's best to raise your issue in that repository.

DoomerDGR8, 2024-07-27:
> That is the repo for the display code your currently running (part of main Opennept4une). We just keep it sperate to make things tidier / easier. It's best to raise your issue in that repository.

I meant do I have to install [display_connector](https://github.com/OpenNeptune3D/display_connector) or not. I'm not the OP. Just curious as since moving to OpenNept4une, I have barely touched the LCD screen and I have the latest firmware on the LCD from the Elegoo Firmware days. So, I suddenly stumble onto this bug report and find out there is a possible new feature for the screen, so I got interesated.

halfmanbear, 2024-07-27:
The main Opennept4une install script can install that repo. Everything is outlined in the wiki instructions.

https://github.com/OpenNeptune3D/OpenNept4une/wiki#post-installation-setup

closed 2024-08-01 by Choccy-vr