OpenNeptune3D/OpenNept4une #200: Error: Option 'pin' is not valid in section 'gcode_macro part_light_off'

BiggumBuddha, 2024-07-15:
Reflashed and updated my N4Max and this gem popped up after reboot. 
[image]

BiggumBuddha, 2024-07-15:
Here's an grab from my current printer.cfg:
[image]

halfmanbear, 2024-07-15:
Have you installed/updated the virtual MCU within `opennept4une` menu option 4?

BiggumBuddha, 2024-07-15:
I did. I just ran through it with the same outcome.

BiggumBuddha, 2024-07-16:
@halfmanbear On a whim, I deleted the generated printer.cfg and recreated it through the opennept4une shell, and it worked correctly. I'll go ahead and close this issue.

closed 2024-07-16 by BiggumBuddha