OpenNeptune3D/OpenNept4une #209: Warping since install

circsmonky, 2024-08-13:
Since I put Opennept4ne on my 4Pro I have been getting lots of warping. I wasn't getting it before. I have tried adjusting many settings, even used a raft and it still warped. I bought an enclosure just to make sure it wasn't a draft that started at the same time, still warped in the enclosure. I really like how opennept4ne works so I didn't want to have to go back to stock, but at it stands my printer doesn't print. I'm sure there's some setting that I need to change but I can't figure out what.

halfmanbear, 2024-08-13:
What slicer were you using before? You can load your old slicer profile and look in our wiki on how to edit the start gcode. 

Then you can see if it's the orca slicer settings causing issue

circsmonky, 2024-08-13:
> What slicer were you using before? You can load your old slicer profile and look in our wiki on how to edit the start gcode.
> 
> Then you can see if it's the orca slicer settings causing issue

I was using orca before, a profile I found on elegoo's discord. I did try that profile a few times but I didn't edit the start gcode, I'll try that tonight and update here.

circsmonky, 2024-08-13:
> What slicer were you using before? You can load your old slicer profile and look in our wiki on how to edit the start gcode.
> 
> Then you can see if it's the orca slicer settings causing issue

i have been hunting all over and i cant find where it tells you how to change the start gcode to match. could you help me locate that?

halfmanbear, 2024-08-13:
https://github.com/OpenNeptune3D/OpenNept4une/wiki/Using-other-Slicers-&-Profiles

circsmonky, 2024-08-14:
> What slicer were you using before? You can load your old slicer profile and look in our wiki on how to edit the start gcode.
> 
> Then you can see if it's the orca slicer settings causing issue

Ok, using my old orca profile it's not warping. Should I just stick to that one? Or try and figure out why?

halfmanbear, 2024-08-14:
Could you export your old profile and upload it here?

circsmonky, 2024-08-14:
[Elegoo Neptune 4 Pro (0.4 nozzle).zip](https://github.com/user-attachments/files/16612687/Elegoo.Neptune.4.Pro.0.4.nozzle.zip)
thats it

circsmonky, 2024-08-15:
> Could you export your old profile and upload it here?

I changed all settings in my opennept4ne profile to match the old one and the ON profile works now. There was definitely something in those settings my printer didn't like. I only did that over just using the other one because the other one doesn't always show up when I open orca.

closed 2024-08-15 by circsmonky