OpenNeptune3D/OpenNept4une #211: update manager issue

wjflier, 2024-08-23:
All of a sudden I'm getting this:  
[image]

halfmanbear, 2024-08-24:
SSH> `opennept4une` and select the option to update the display usually menu option No# 5

wjflier, 2024-08-24:
Did that, no joy

halfmanbear, 2024-08-24:
Can you upload your moonraker.conf

wjflier, 2024-08-24:
[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /home/mks/printer_data/comms/klippy.sock

[authorization]
trusted_clients:
    10.10.0.0/16
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
cors_domains:
    *.lan
    *.local
    *://localhost
    *://localhost:*
    *://my.mainsail.xyz
    *://app.fluidd.xyz

[file_manager]
enable_object_processing: True

[octoprint_compat]

[history]

[update_manager]
channel: dev
refresh_interval: 168

[update_manager fluidd-config]
type: git_repo
primary_branch: master
path: ~/fluidd-config
;origin: https://github.com/fluidd-core/fluidd-config.git
managed_services: klipper

[update_manager fluidd]
type: web
channel: stable
repo: fluidd-core/fluidd
path: ~/fluidd

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

[update_manager Klipper-Adaptive-Meshing-Purging]
type: git_repo
channel: dev
path: ~/Klipper-Adaptive-Meshing-Purging
origin: https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging.git
managed_services: klipper
primary_branch: main

[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
managed_services: crowsnest
install_script: tools/install.sh

[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
env: ~/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install.sh

[update_manager OpenNept4une]
type: git_repo
primary_branch: main
path: /home/mks/OpenNept4une
is_system_service: False
origin: https://github.com/OpenNeptune3D/OpenNept4une.git

[update_manager display]
type: git_repo
primary_branch: main
path: /home/mks/display_connector
virtualenv: /home/mks/display_connector/venv
requirements: requirements.txt
origin: https://github.com/OpenNeptune3D/display_connector.git

[power tasmota_plug]
type: tasmota
address: 192.168.50.224
password: ******

[include octoeverywhere-system.cfg]

[include moonraker-obico-update.cfg]

wjflier, 2024-08-24:
Just realized we're on discord too

alldone, 2024-08-31:
same problem, and if i update from shell ( openneptune--->5-->y) have this error:

[image]

halfmanbear, 2024-08-31:
You shouldn't ever log in as root, only mks

alldone, 2024-08-31:
tnx it worked,

now after start a new print, it calculate adaptive bedmash and then go in pause state....

closed 2024-09-13 by halfmanbear