OpenNeptune3D/OpenNept4une #214: MCU Error after Update

ElektroCoder, 2024-09-13:
After the update, I suddenly receive the following error when setting the nozzle temperature to 220°C. Previously, the printer worked absolutely fine:

MCU 'mcu' shutdown: Rescheduled timer in the past This generally occurs when the micro-controller has been requested to step at a rate higher than it is capable of obtaining. Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Printer is shutdown

halfmanbear, 2024-09-13:
https://klipper.discourse.group/t/timer-too-close-so-far-two-prints-failing-on-the-same-layer-wtf/18782/4

Bug in latest klipper, the FIX is 

```
sudo service klipper stop
cd ~/klipper
git checkout 293858c51fc51c998a046ba12c00da0418fb8403
sudo service klipper start
```

ElektroCoder, 2024-09-13:
Thank you so much, @halfmanbear! You’ve truly made my day with your solution. I was struggling with this issue, and your fix saved me a lot of trouble. I really appreciate it!

closed 2024-09-20 by halfmanbear