OpenNeptune3D/OpenNept4une #217: Input shaping and Fan control

Bad-Hackerman, 2024-09-16:
Is there a way to do input shaping on openNept4une?
Haven't found any documentation or setting/macro

I have 4 Fans I can control through the fluidd interface but only the part fan works, but it acts on the aux fan as well

lexxxel, 2024-09-27:
Read the Klipper docu about input shaping and resonance measurement: https://www.klipper3d.org/Measuring_Resonances.html#measuring-the-resonances_1

The part fan and the aux fan are on the same pin, meaning eleggoo's hardware layout does not allow to control both fans independently.

Also your asking for support and do not report an issue with this project. Please use the discussion tab.

halfmanbear, 2024-10-17:
I'm currently working on a fix for the SPI ADXLs on the Plus and Max. Ive just been donated a V2.0 board so I can debug the issue

jimmycr, 2024-10-24:
If it will be possible to make it work for Plus and Max version using built-in ADXL accelerometers, it will be huge improvement for OpenNept4une. Thank you for your effort @halfmanbear !!

halfmanbear, 2024-10-24:
If you join our discord and check out this thread. I've been debugging it today.

https://discord.gg/4bw3MwVv

https://discord.com/channels/1206903884124004413/1266720385823670282/1298950195043635221

closed 2025-03-10 by halfmanbear