OpenNeptune3D/OpenNept4une #233: fly-adxl345

Entasdisk1, 2024-11-26:
hello
i bought fly-adxl345 for my neptune 4 printer.
i am trying to install it according to the instructions in the wiki but it gives an error.
i can't write from discord because it is blocked in my country.
i wrote the error i got:
[printer.txt](https://github.com/user-attachments/files/17920628/printer.txt)

Requirement already satisfied: numpy in /home/mks/klippy-env/lib/python3.11/site-packages (1.26.2)

[notice] A new release of pip is available: 23.3.2 -> 24.3.1
[notice] To update, run: /home/mks/klippy-env/bin/python -m pip install --upgrade pip
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
src/neopixel.c:50:10: fatal error: avr/interrupt.h: No such file or directory
   50 | #include <avr/interrupt.h> // TCNT1
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:65: out/src/neopixel.o] Error 1
  Compiling out/src/neopixel.o
src/neopixel.c:50:10: fatal error: avr/interrupt.h: No such file or directory
   50 | #include <avr/interrupt.h> // TCNT1
      |          ^~~~~~~~~~~~~~~~~

robertoamd90, 2025-01-16:
same problem here,

any solution?

halfmanbear, 2025-01-16:
ive posted a possible fix to the dev branch (you can switch branches within `OpenNept4une.sh`

https://github.com/OpenNeptune3D/OpenNept4une/commit/c5a2fe40587527b98bca935707199b3e7dc321c6

halfmanbear, 2025-01-20:
Yes that should do the trick

R1D3R175, 2025-02-08:
Switching to the `dev` branch using the advanced menu and then updating the ADXL firmware does indeed work

robertoamd90, 2025-02-16:
perfect, last update fix the problem in the main branch!

closed 2025-03-10 by halfmanbear