OpenNeptune3D/OpenNept4une #240: MCU Protocol Error

Evan1719, 2024-12-16:
After flashing my Neptune 4 pro with OpenNept4une. I am receiving an MCU Protocol error on my Fluidd dashboard. The MCU's were updated properly as the file was changed to cursor. What is the issue occurring?

_Originally posted by @Evan1719 in https://github.com/OpenNeptune3D/OpenNept4une/discussions/239_

halfmanbear, 2024-12-16:
Do you have a screenshot of the full error message? Also could you upload your klippy.log

Ensure you installed our printer cfg in the opennept4une main installer.

cvopicka, 2024-12-16:
I ran into this last night.  It had to do with the accelerometer information in printer.cfg.  I have a different/no accelerometer.  I commented out those 2 sections and no more errors.  Be warned my machine is not fully running yet.  Only got it installed.  Hope to test tonight.  So all I know right now is that gets rid of the error.

Tonyzaa11, 2024-12-18:
I'm having the same issue after a fresh install.
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-401-g383b83d78-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-401-g383b83d78
rpi: Current version v0.12.0-401-g383b83d78
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'rpi': Unknown command: query_adxl345

[image]

Johnewb, 2024-12-19:
I also just did a fresh install and am having the same issue, 
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-401-g383b83d78-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-401-g383b83d78
rpi: Current version v0.12.0-401-g383b83d78
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'rpi': Unknown command: query_adxl345

[klippy.log](https://github.com/user-attachments/files/18191048/klippy.log)

when I do the KIAUH updates it also fails to complete the system updates,
[image]

SnprSlick, 2024-12-19:
Just did an install on my neptune4max and have the same issue.

Tonyzaa11, 2024-12-19:
okay, the weirdest thing happened, this is NOT a fix/workaround, but I reverted back 1 commit in kiauh and re-updated again, in fluidd settings > update ALL, and everything is working.
it doesn't make any sense tho

54iceblock, 2024-12-20:
I had the same issue today.  Committing out the two blocks associated with "adxl345" in printer.cfg worked for me.

adelyser, 2024-12-22:
I had the same issue on my Neptune 4 Max. From ssh, I navigated to the klipper directory and ran the following:

```
make clean
make menuconfig
configured for Linux process under Microcontroller Architecture
make
sudo systemctl stop klipper
make flash
sudo systemctl start klipper
```

And that fixed my issue. It appears there is an issue with the script and how it installs the host klipper service. I do have adxl's installed, so those errors are gone after restarting klipper.

adelyser, 2024-12-22:
It appears to be an issue with the config used to compile the `virtual rpi` device. Running a diff between the latest klipper config after running `make menuconfig` and the config used by the OpenNept4une config:

```
diff OpenNept4une/mcu-firmware/virtualmcu.config klipper/.config
8c8
< # CONFIG_MACH_RP2040 is not set
---
> # CONFIG_MACH_RPXXXX is not set
21c21,22
< CONFIG_WANT_SENSORS=y
---
> CONFIG_WANT_THERMOCOUPLE=y
> CONFIG_WANT_ADXL345=y
22a24,28
> CONFIG_WANT_MPU9250=y
> CONFIG_WANT_HX71X=y
> CONFIG_WANT_ADS1220=y
> CONFIG_WANT_LDC1612=y
> CONFIG_WANT_SENSOR_ANGLE=y
```

halfmanbear, 2024-12-23:
Dev branch has a fix, once approved will be in main. Thankyou for raising this!

closed 2025-01-16 by halfmanbear