OpenNeptune3D/OpenNept4une #243: printing on the back half of bed

shackleford92, 2024-12-26:
i am honestly not sure whether i have fudged the printer or an orcaslicer setting.
printer: neptune 4 pro
steps for ME to reproduce:  set an object centered on the build plate in orcaslicer, upload, and print
expected outcome: object to print as displayed in slicer software
issue incurred: object is centered on the x axis but prints on the back 3/4 of they axis
what i believe may be relevant information: printer homes to x:141.75   y:97.05    z:10(the asymmetrical x,y raises flags in my mind),       it seems to print as expected sometimes?,   i have been dealing with this for a while but i do small prints and have not run into too many issues,       the homing seems to be a printer thing and not slicer?,     i have not directly edited printer.cfg MUCH    the last thing i did was  add and comment out resonance includes
#[include adxlX.cfg]             ; Comment this out when you disconnect the Pico/MellowFly      accelerometer mounted to print head
#[include adxlY.cfg]             ; Comment this out when you disconnect the Pico/MellowFly      accelerometer mounted to print bed
i do not remember it being an issue at that time in the july timeframe,
maybe an orcaslicer upgrade migration issue?

shackleford92, 2025-01-17:
sooo, not sure if this was from bit corruption from an outage BUT those asymmetrical homing coordinates were baked into my printer.cfg

[safe_z_home]
home_xy_position: 141.75,97.05
speed: 100
z_hop: 10                 
z_hop_speed: 5

making these alterations seems to have resolved my issue

[safe_z_home]
home_xy_position: 117.5,117.5
speed: 100
z_hop: 10                 
z_hop_speed: 5

unsure if this came from fat fingering, source, corruption, a bad git pull, who knows 

may be worthwhile to commit @halfmanbear

halfmanbear, 2025-01-17:
https://www.klipper3d.org/Config_Reference.html?h=pixel#bed_mesh

"When using a probe to home the z-axis, it is recommended to define a safe_z_home section in printer.cfg to home toward the center of the print area."

Due to the inductive probe being offset from the print nozzle, the homing coordinates are the centre of the bed with the sensor offset applied to X and Y. This ensures the probe is homing in the very middle of the bed.

shackleford92, 2025-02-16:
i had been gaslit by myself

the problem persisted
i eventually reran input shaper, i used ResHelper (great repo, suggest including it in this repo), idk what it did to resync the steppers or what, someone more knowledgeable than me can dig through the code, BUT MY PRINTS HAVE BEEN EXACTLY WHERE EXPECTED SINCE RUNNING RESHELPER.

closed 2025-03-10 by halfmanbear