OpenNeptune3D/OpenNept4une #245: Problem with Wi-Fi connection

RomanSych55, 2025-01-06:
После устнавки OpenNept4une при подключении к нему по сети Wi fi постоянно просиходит отключение, такая же проблема при отправки файлов из слайсера на печать. Помогает только перезагрузка хоста

halfmanbear, 2025-01-13:
Какой адаптер Wi-Fi вы используете? Встроенный (модель плюс или макс) или внешний (модель 4/про)?

RomanSych55, 2025-01-13:
Встроенный модуль Neptune 4 Max. При подключении через LAN порт к Wi-Fi
роутеру все работает стабильно.

eternal-sorrow, 2025-02-20:
У тебя в логах случаем нет таких сообщений перед дисконнектом:
```
[dhd] arpoe_enable not set
[dhd] dhd_process_pkt_reorder_info: received flags to cleanup, but no flow (14) yet
[dhd] dhd_process_pkt_reorder_info: received flags to cleanup, but no flow (15) yet
```

Числа в скобках постоянно инкрементируются.