OpenNeptune3D/OpenNept4une #246: Uploading of large files abruptly stops and presents a 502 Bad Gateway nginx/1.22.1 error

robertjtaggartjr, 2025-01-10:
When trying to upload large gcode file (644,824KB) to the printer the progress bar appears and gets to 100% but then it presents this error502 Bad Gateway nginx/1.22.1 

I have increased the timeout in the nginx conf file. 
I Have tried to upload via USB but the printer doesn't seem to read the USB sticks anymore .. Unless I mount the drive manually 

I have made sure all the services are up to date 

[image]

halfmanbear, 2025-01-13:
Do you have sufficient storage available on your machine, also are you uploading directly via orcaslicer?

robertjtaggartjr, 2025-01-13:
Yes I have updated the MMU to 30gigs.  I have also tried via Orca and via Klipper both with same results 
[image]

robertjtaggartjr, 2025-01-15:
I was able to upload a file of 212.45MB with no issues..  It just seems to be larger files.

dimagre, 2025-01-18:
The same problem here, after updating all the software

halfmanbear, 2025-02-05:
You can try the following for a potential solution 

`sudo nano /etc/nginx/nginx.conf`

Add the following within the http section (ensure you press tab to indent each line).

```
    proxy_read_timeout 600;
    proxy_send_timeout 600;
    client_body_timeout 600;
```

Press `Ctrl+X` then `y` to save and exit.

Reboot with 

`sudo reboot`

closed 2025-03-10 by halfmanbear