OpenNeptune3D/OpenNept4une #249: can't open Opennept4tune

mbscomp16, 2025-01-14:
ran this command. cd ~/OpenNept4une/ && git remote set-url origin https://github.com/OpenNeptune3D/OpenNept4une.git && git fetch --all && git reset --hard origin/main && git clean -fd. fails with an error. fatal: not a git repository (or any of the parent directories): .git. I used mkdir Opennept4une and cd in the new directory. ran the same command. received the same error,

halfmanbear, 2025-01-14:
You've flashed your eMMC correctly?

On a fresh install of the Linux image, you must log in as user mks not root.

Typing `ls -al` after login should show an Opennept4une directory already present.

mbscomp16, 2025-01-14:
currupted image file Neptune4-Plus+Max-v0.1.5-ZNP-K1-V2.0.img.xz. every downloaded file is corrupt. use balena etcher url link. still currupt had to use Neptune4-Plus+Max-v0.1.4-ZNP-K1-Vx.x.img.gz. after installing and updating. i still can not install OpenNept4une git. cd ~/OpenNept4une/ && git remote set-url origin https://github.com/OpenNeptune3D/OpenNept4une.git && git fetch --all && git reset --hard origin/main && git clean -fd. fails not a valid GitHub.  ls -al does not show Opennept4tune.  this is my 4th fresh install. I will try again tonight

mbscomp16, 2025-01-14:
ssh mks@ip. no root login.

mbscomp16, 2025-01-14:
fresh install of Neptune4-Plus+Max-v0.1.4-ZNP-K1-Vx.x.img.gz. ssh mks@ip. mks@mkspi:~$ mks@mkspi:~$ ls -al Opennept4une Directory is not present.

mbscomp16, 2025-01-14:
from the home directory. I ran git clone --depth=1 -b main https://github.com/OpenNeptune3D/OpenNept4une.git. I followed the rest of the instructions. install is working. thank you

closed 2025-01-16 by halfmanbear