OpenNeptune3D/OpenNept4une #252: Filament runout issue

dj050894, 2025-01-21:
Ran into an issue today where I was unable to resume print after filament runout. The print paused as it should when the sensor was triggered, I was out of the house when it went off. When I got back to the printer after about an hour and a half the bed heater had turned off and cooled to room temp. Print was obviously ruined at this point, but I swapped the filament and hit resume to see what would happen. The bed and nozzle did not reheat and print never resumed.

halfmanbear, 2025-01-21:
There is a 12h safety timeout on pause, can you check this in your printer.cfg PAUSE macro

`SET_IDLE_TIMEOUT TIMEOUT=43200`

Also check in your fan section you have 

`idle_timeout: 43200`

dpinse, 2025-01-22:
I've been chasing this issue as well, it seems that the printer locks up when the filament sensor is triggered, the print head fan ramps to full speed and the printer becomes unresponsive from both the touchscreen and via network, the bed and nozzle also go cold and only a hard power cycle get it out of this state. Presently I have the sensor unplugged as a work around.

halfmanbear, 2025-01-22:
Under the section 

`[filament_switch_sensor filament_sensor]` in your `printer.cfg` can you change the value `pause_delay:` from `1.0` to `10.0` and test the filament runout procedure.

dpinse, 2025-01-22:
No luck same behavior, the frame light flickers and turns off right before it locked up as well

halfmanbear, 2025-01-23:
Can you attach your latest klipper log on here please?

dpinse, 2025-01-23:
[klippy.log](https://github.com/user-attachments/files/18524884/klippy.log)

Attached, it seems to only happen when the filament run out sensor is triggered. Even just removing the filament from the sensor when the printer is idle causes the crash

halfmanbear, 2025-02-05:
When was the last time you updated the printer.cfg within `opennept4une.sh` it appears to be an old version you are running.

Also check `user_settings.cfg` and make sure there is no additional code regarding runout or pause etc