OpenNeptune3D/OpenNept4une #254: Neptune4 max / flashing / adxl345 / Error

akugiz, 2025-02-02:
"Hey, can somebody help me? When I'm flashing the Neptune4 Max, I get an error. Out of five attempts, it only works once. However, in Fluid, I keep getting this: 'MCU Protocol error.'

When I remove the entire [adxl345 x] and [adxl345 y] sections from the printer config, everything works fine.

Where could the problem be? I was thinking maybe the accelerometer isn't flashing correctly?

I uploaded all the screenshots."

**This I'm removed from printer.config**
[image]

**Error when I flash MCU** 
[image]

**Fluid Error**
[image]

halfmanbear, 2025-02-05:
Can you run `htop` in the terminal and work out what is utilising all your RAM.

Also just check you have enough storage space fluidd as a good measure.

Have you tried the microSD flash of the main MCU? This is still showing as elegoos original version.

akugiz, 2025-02-05:
> Can you run `htop` in the terminal and work out what is utilising all your RAM.
> 
> Also just check you have enough storage space fluidd as a good measure.
> 
> Additionally try switch to the dev branch of opennept4une (advanced page within opennept4une.ch) to see if you can flash your adxl
> 
> Have you tried the microSD flash of the main MCU? This is still showing as elegoos original version.

Hey, thanks for the reply. Yes, after this line: 'Linking out/klipper.elf,' the system seems to crash, using 100% CPU. It happens just after a few attempts to flash the files. Everything is resolved now, but I just need everything to be separate, like how 'STM32' and 'Virtual Pi' are handled separately. What could be causing that crash?

I'm attaching a video of what’s happening.

<div style="padding:56.25% 0 0 0;position:relative;"><iframe src="https://player.vimeo.com/video/1053890272?h=a536c053fb&amp;badge=0&amp;autopause=0&amp;player_id=0&amp;app_id=58479" frameborder="0" allow="autoplay; fullscreen; picture-in-picture; clipboard-write; encrypted-media" style="position:absolute;top:0;left:0;width:100%;height:100%;" title="2025-02-05 19-34-59"></iframe></div><script src="https://player.vimeo.com/api/player.js"></script>

musicCodeGuy, 2025-06-30:
I've had this exact same problem several times on a newish neptune4plus (bought late last year). I had the mcu firmware build once after adding a swap file, but had kernel panic after kernel panic and went back to the stock firmware. I think there may be some newer board revisions that don't quite work with the opennept4une images.

closed 2026-10-30 by halfmanbear