OpenNeptune3D/OpenNept4une #255: [Feature Request]

bolens, 2025-02-02:
Similar to the fantastic [Creality Helper Script](https://github.com/Guilouz/Creality-Helper-Script-Wiki) by Guilouz, I think it would be a great addition to add auto printer config backup to GitHub.  The workflow is roughly along the lines of configuring a repo and login in the script, then every time you save your printer config it will auto-push to the repo for easy restoring, reverting, and sharing of your config.

halfmanbear, 2025-02-05:
Just linking this here for future reference 

https://guilouz.github.io/Creality-Helper-Script-Wiki/helper-script/git-backup/?h=git

TheMeanCanEHdian, 2025-03-27:
Something like [Klipper-Backup](https://klipperbackup.xyz/) might be better here. And it can be installed and configured with [KIAUH v6](https://klipperbackup.xyz/installation/#2-kiauh).

closed 2026-10-30 by halfmanbear