OpenNeptune3D/OpenNept4une #261: Initial filament purge is happening underneath support base area

voska, 2025-02-19:
I'm not sure if this is the right place to post this, but I'm having an issue where the support base is being printed on top of the purge line. I'm running the latest OpenNept4ne on my N4+ 

Problematic file: https://www.printables.com/model/815360-neptune-4-spool-roller-mod-with-608-bearings-rotat

[image]

Purge line position shown in blue:

[image]

Ataraxia-Mechanica, 2025-02-25:
This is a problem with orca slicer, its exported printing area doesn't include support, brims and prime tower.

halfmanbear, 2025-02-26:
You can edit / increase the smart purge line distance in kamps configuration file within fluidd

closed 2025-02-26 by halfmanbear