OpenNeptune3D/OpenNept4une #264: Unable to proceed beyond flashing/updating MCU - 502 bad gateway

Barkingmad15, 2025-02-22:
Hi, I would be very grateful for some assistance please. I am attempting to install the latest release: [v0.1.5-ZNP-K1-24.02.0-current-6.1.67](https://github.com/OpenNeptune3D/OpenNept4une/releases/tag/v0.1.5) onto a Neptune 4 Plus. I have sucessfully written the image & all steps have gone flawlessly until it comes time to log into http://my_printer_ip/#/configure/ At this point I get a Moonraker warning / Nginx bad gateway. I get asked to add a printer but if I put the printers IP address in I get "blocked by CORS policy" I have tried everything I can think of using http://fluid.local or localhost or 127.0.0.1 or even using the LAN port instead of WiFi. Those efforst result in bad gateway from Nginx. Only the actual WiFi IP address results in blocked by CORS. 

I can login using WinSCP & copy the files manually to the micro SDcard BUT that does me no good as I cannot access the Fluiid interface in later steps. I even retried flashing over from the start but get the same error. I checked my board version & it's the V2.0 as it should be in the Plus. I use Klipper on a Neptune 3 Plus as well, so I'm familiar with all the basics, I just don't know how to get past this error at this stage. I would be grateful for guidance please & thank-you.

[image]
[image]
[image]

FadingCaptcha, 2025-02-23:
i am having the same issue. im on neptune4pro 1.0 board 1.2 steppers.

IronicMollusk, 2025-02-23:
Same issue here on the neptune 4 max, using the 0.1.5 release

bwaslo, 2025-02-23:
Looks like same problem here.  N4pro 1.0 board, 1.2A steppers.  A bunch of Moonraker warnings afterwards.

Also I can't get crowsnest configured for usb camera, it asks me how many cameras and after I reply "1" it starts with something like "sorry that you decided to uninstall crowsnest, are you sure you want to?".  And since I didn't ask to uninstall I reply "N" and then it throws an error.  Frustrating evening.

svibb, 2025-02-23:
Getting the same issue with my Neptune 4 Max and latest Nept4une img.

[image]

Henkeh, 2025-02-23:
I think I had similar issues when I first installed it. If memory serves, I fixed it by added my home domain under trusted_clients in the moonraker.conf file, e.g.:

[image]

Of course change the 1 in 192.168.1.0/24 to the right IP range you chose for your home domain.

IronicMollusk, 2025-02-23:
@Henkeh, thats a good point. here's my current settings for moonraker, or at least the authorization portion:
[authorization]
trusted_clients:
    10.10.0.0/16
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
    
cors_domains:
    *.lan
    *.local
    *://localhost
    *://localhost:*
    *://my.mainsail.xyz
    *://app.fluidd.xyz

I am on a slightly different subnet then this includes. I'll update the settings in the morning and see if that works. thanks for the heads up.

svibb, 2025-02-23:
I figured it out, looks like the link is broken to download and install moonraker through kiauh. I manually downloaded and installed moonraker from source and then updated through kiauh normally. I reran all the install configs through opennept4une and got access to moonraker afterwards. Getting the mcu updated right now.

Barkingmad15, 2025-02-23:
> I figured it out, looks like the link is broken to download and install moonraker through kiauh. I manually downloaded and installed moonraker from source and then updated through kiauh normally. I [...]

This was the answer. Thank-you. Followed those steps & it's flashing now 😁 Also, thanks to @Henkeh tried that but it didn't work. As an addon to that if you follow the steps to download Moonraker independently, you'll need to remove that edit from Moonraker.conf It doesn't play nice.

closed 2025-02-23 by Barkingmad15

bwaslo, 2025-02-23:
"I figured it out, looks like the link is broken to download and install moonraker through kiauh. I manually downloaded and installed moonraker from source and then updated through kiauh normally. I reran all the install configs through opennept4une and got access to moonraker afterwards. "

Can someone be more explicit (i.e., give actual commands/steps/instructions) for doing this, like in the OpenNept4une installation instructions?  I'm not a Linux or Python programmer and can't figure out what exactly the above is telling me to do!

Barkingmad15, 2025-02-23:
> "I figured it out, looks like the link is broken to download and install moonraker through kiauh. I manually downloaded and installed moonraker from source and then updated through kiauh normally. I [...]

Sure! I don't know if there is a right or wrong way to do this, but for me, I visited this site: https://moonraker.readthedocs.io/en/latest/installation/ Scroll down to the commands to install from source & entered them individually. Interestingly enough it didn't work as an all in one command, so I split it up. That installed the base Moonraker package & after that I just went back to Kiauh & checked for updates, but there weren't any. I checked at this point that the web interface was working & it was, so carried on with the install, which I am glad to say completed sucessfully.

bwaslo, 2025-02-23:
Thanks.  But what does "back to Kiauh" mean?  Kiauh, if I'm remembering correctly was a word that flew by during the installation attempt, I don't recall starting something related.  (Sorry to be so dense)

Barkingmad15, 2025-02-23:
> Thanks. But what does "back to Kiauh" mean? Kiauh, if I'm remembering correctly was a word that flew by during the installation attempt, I don't recall starting something related. (Sorry to be so dense)

No, that's OK. Sorry if I was a bit vague, just get wary of putting too much instruction.  Kiauh is an installation assist for Klipper & all it's accompanying services, such as Moonraker, Fluid, Mainsail etc. The guts who put OpenNept4une together already have it as part of the image, so you can just start the command, as per the wiki instructions, without having to worry about cloning the repo etc. You can see this link for more info on Kiauh https://github.com/dw-0/kiauh Hope this helps.

FadingCaptcha, 2025-02-23:
> Thanks. But what does "back to Kiauh" mean? Kiauh, if I'm remembering correctly was a word that flew by during the installation attempt, I don't recall starting something related. (Sorry to be so dense)

I just reflashed and tried again. went over to the Kiuah issues on github and it seems there was an issue with downloading moonraker from the version of kiuah we were trying yesterday. https://github.com/dw-0/kiauh/issues/647. try updating kiauh (this should be the "**Update pre-loaded plugins**" step on openeptune wiki , mine is finally working now.

bwaslo, 2025-02-24:
Thanks everyone, that seems to have it sorted, no red error messages now at least and webcam runs in fluid (though not in mainsail.... But that's ok.  Next to install Cartographer3D and see if I can get that running