OpenNeptune3D/OpenNept4une #270: Bed Grid no longer displayed on 4 Max after update

SlotCertifier, 2025-03-08:
Recently applied updates via Mainsail and now I no longer get display of the bed grid during the adaptive mesh.

halfmanbear, 2025-03-10:
Thank you for raising this. I believe it's due to this recent klipper update;

https://github.com/Klipper3d/klipper/commit/edc3d34beb10448be518d20c9e55421638776a75

In our `display_connector` repo, we were looking for the following string `("// bed_mesh: generated points")` to initiate the display of the bed probing grid.

I've created a fix for this in the dev branch of `display_connector`.
https://github.com/OpenNeptune3D/display_connector/commit/8c5e18a91d286b54adab58e30ec0e524c1d712a2

Once I've tested that it doesn't break any other functionality I'll push it to `main`.

SlotCertifier, 2025-03-10:
Appreciate your fast response.

Yahoo Mail: Search, Organize, Conquer

closed 2025-03-10 by halfmanbear