OpenNeptune3D/OpenNept4une #276: Extruder overheating

dotstal, 2025-03-20:
Extruder overheating
When printing starts, the extruder overheats by 2-3 degrees, then it cools down, and then printing starts.
Is it possible to start printing with an overheated extruder? During the test run - it will cool down to the required temperature.
Or add this correction in the next releases

halfmanbear, 2025-03-20:
Hello,

We already have this configured in the current print start macro.

`TEMPERATURE_WAIT SENSOR=extruder MINIMUM={EXTRUDER_TEMP-4} MAXIMUM={EXTRUDER_TEMP+10} `

This will allow the print to begin 4°C under target and up to 10°C over target.

If your extruder is overheating the target temp by 10°C, I  recommend performing the PID tune macro for your extruder, this will reduce any overshoot.

dotstal, 2025-03-20:
Thanks, it could increase to 8
`MAXIMUM={EXTRUDER_TEMP+8}`

closed 2025-03-20 by dotstal

halfmanbear, 2025-03-20:
That would be a reduction in the allowed overshoot by 2°C as it's already set to +10°C.

Current config allows the print to start (for example ) if your target extruder temp is 220°C the print will start if the extruded is beti 216°C - 230°C .