OpenNeptune3D/OpenNept4une #278: Inconsistent Z-Offset on Subsequent Prints – Requires Full Reboot on Neptune 4 Plus

fabiogfabio, 2025-03-26:
Description
I'm experiencing an issue with my Elegoo Neptune 4 Plus running the OpenNeptune firmware. The first print after powering on the printer comes out perfectly, but if I print the same object again without a full power cycle, the top layers are noticeably squashed. It appears that the Z-offset drops between prints, even though the first layers remain correct.

Steps to Reproduce
Power on the printer.

Run a print using the following configuration:

Adaptive Bed Mesh Calibration is enabled via the printer configuration (BED_MESH_CALIBRATE ADAPTIVE=1).

I calibrate the probe offset manually using CALIBRATE_PROBE_Z_OFFSET and save it with SAVE_CONFIG.

The slicer start G-code does not include any SET_GCODE_OFFSET commands.

The first print is perfect.

Without rebooting the printer, run the same print again.

Observe that the top layers are printed too low (indicating a drop in Z-offset).

Expected Behavior
The printer should maintain the correct Z-offset and bed mesh calibration between prints without requiring a full restart.

Observed Behavior
First Print (After Reboot): Perfect print with correct Z-offset.

Subsequent Prints (Without Rebooting): Top layers are squashed, as if the Z-offset has decreased.

Troubleshooting Attempts
Removed any SET_GCODE_OFFSET commands from the slicer's start G-code.

Used adaptive bed mesh calibration in the printer configuration.

Calibrated the probe offset with CALIBRATE_PROBE_Z_OFFSET and saved with SAVE_CONFIG.

Removed FIRMWARE_RESTART from the stop G-code.

Verified temperature stability and hardware connections.

Tested with both adaptive and fixed bed mesh methods.

None of these approaches prevented the Z-offset from dropping on subsequent prints.

Environment
Firmware: OpenNeptune (latest version)

Printer: Elegoo Neptune 4 Plus

Slicer: OrcaSlicer (using custom macros for PRINT_START/PRINT_END)

Setup: The first print (after a full reboot) is always perfect. Subsequent prints (without reboot) consistently show squashed top layers.

Attachments
Image 1 (Correct print after reboot): Printer On - Correct Print 
[https://imgur.com/w91PiyM](url)
Image 2 (Print with squashed top layers after subsequent print): Subsequent Print Issue
[https://imgur.com/a/txVXVEo](url)

Conclusion
It seems that after the first print, certain internal parameters—possibly the Z-offset or the adaptive bed mesh calibration—are being reset or modified. This forces me to reboot the printer between prints to achieve consistent results.

Any help or insights on whether this is a known issue with OpenNeptune or suggestions for a workaround would be greatly appreciated.

Thank you!

cdichiara, 2025-04-09:
You know, I might not have had the same issue as you, but I had to replace my Nyxlab Auto Leveling Sensor for my Neptune 4 Plus. It was so inaccurate that the bed leveling was failing, it couldn't reliably get three measurements from the same location that were consistent enough. My z offset was constantly varying from print to print. I got a replacement and my issues have gone away.

halfmanbear, 2025-05-05:
How are you powering off / rebooting your printer?
Also in fluidd when saving your offset make sure to refresh the page in the browser before clicking the small save icon to the right of the live zoffset adjust. Same goes for an orange `save config` alert at the top of fluid.

Secondly it's worth viewing your printer.cfg before and after a saved change. The actual zoffset value will be stored in the grey area at the bottom of your printer.cfg it ignores the one further up the config. 

Finally ensure no zoffset is set within user_settings.cfg

fabiogfabio, 2025-05-05:
Thanks for the suggestions!

To answer your question: I’m powering off the printer completely via the main switch — full shutdown, not just a restart from Fluidd or via FIRMWARE_RESTART.

Here's what’s happening:

First print after powering on is perfect — great first layer, no issues.

I remove the part and immediately start the exact same print again (tested both from Fluidd and from the printer’s panel).

Now, the first layer is messed up — it looks like the Z offset is off, or the mesh isn't being applied properly.

If I power cycle the printer again and reprint, it works flawlessly once more.

I’ve tried both adaptive and standard mesh modes, reloaded and saved config, restarted Klipper, even refreshed the browser before saving offset changes. The issue persists until I do a full power off.

Also double-checked: no Z offset is being set in user_settings.cfg, and the correct value does appear in the lower section of printer.cfg after saving.

That’s why I’m stuck — it behaves like something in memory (maybe related to the mesh or Z offset) only resets fully after a hard power cycle.

Any idea what else I might check?

fabiogfabio, 2025-05-07:
[https://i.imgur.com/AdgmnFe.jpeg
https://i.imgur.com/qrTdgGG.jpeg](url)

halfmanbear, 2025-05-10:
Your printer.cfg is quite outdated.

Could you SSH into the printer and run the command `opennept4une`.

Accept any updates that appear, then confirm the correct model of printer is listed under the opennept4une logo at the top of the menu.

Select `1) Install/Update OpenNept4une printer.cfg`

This will keep your previous bed mesh, PID tune and z-offset calibrated values.

fabiogfabio, 2025-05-10:
[image]

It's not normal that the real model doesn't appear to me

halfmanbear, 2025-05-10:
You can re-set your model if it's not displaying within the advanced operations page

fabiogfabio, 2025-05-10:
The issue still persists even after the update. The exact same behavior occurs—after a full reboot, everything starts working again.

halfmanbear, 2025-05-12:
The screw adjustment values are comparisons of probe trigger heights.

The z-offset / probe calibration value is the calibrated offset between (probe to bed) and (nozzle to bed) measurements.

https://github.com/OpenNeptune3D/OpenNept4une/wiki/Printer-Calibration-%E2%80%90-Klipper-&-OrcaSlicer#more-info-on-z_offset

fabiogfabio, 2025-05-12:
Hi,

Thanks for the reminder — I am already following the correct shutdown procedure outlined in the wiki.

I understand the distinction between screw adjustment values (related to probe trigger height comparisons) and the Z-offset, which represents the calibrated difference between the probe-to-bed and nozzle-to-bed measurements.

What I’m struggling to understand, though, is why the first print after powering on the printer comes out perfectly, but the second one fails unless I perform a full reboot.
It’s as if something in the calibration or state of the printer changes — or isn't properly retained — between prints.
Could it be that some initialization or calibration routine only runs correctly after a fresh start? Or maybe a value like Z-offset isn't being re-applied correctly for subsequent prints?

Any insights into what might be causing this behavior would be really appreciated.

halfmanbear, 2025-05-12:
I don't think it's a software issue, if it's not changing in the printer.cfg then it's fixed.

Have you tried increasing the bed heat soak time before your first print? It may just be the bed changing shape due to being fully thermally soaked by the 2nd print?

https://github.com/OpenNeptune3D/OpenNept4une/wiki/Printer-Calibration-%E2%80%90-Klipper-&-OrcaSlicer#plus--max-heat-soak

fabiogfabio, 2025-05-12:
I see your point, but I don’t believe it’s a software issue either. The configuration in printer.cfg seems to remain fixed, as you mentioned.

Regarding the bed heat soak, I understand your suggestion, but the issue persists even with different prints. For example, Print 1 is fine, but Print 2 fails, and this happens consistently. If I power off and on the printer and immediately launch another print, Print 3 comes out fine again, and this cycle repeats itself — it doesn’t seem to be related to the time gap between prints.

I’ve tried variations in bed soak times, but the issue continues to occur regardless of the elapsed time between prints. It almost feels like something is reset after a reboot, but without a change in the heat soak or other parameters.

Any further thoughts on what might be happening?

halfmanbear, 2025-05-12:
The only time ive come across something unusual like this was when someone had a loose / miss-seated bed level sensor on the print head. 

Does the issue disappear if you were to insert your original elegoo eMMC / revert to their firmware?

fabiogfabio, 2025-05-12:
Just a few weeks ago, I replaced the entire extruder assembly because the gear was slipping, but the problem still persists, so I would rule out any sensor issues—since the previous one had the same problem as well.
To be honest, I haven’t tried reverting back to the stock firmware yet.

OlsMobile, 2025-06-05:
Same printer, similar issue : first print runs well, then i have to manually increase Z-offset +0.005/+0.010 each print, first on stock firmware, then with OpenNeptune. On power off or reboot, i've to re-adjust z-offset or set manually the last value +0.005. Tried many things, same as fabio, but nothing changed

fabiogfabio, 2025-06-26:
Hi everyone!
I’ve swapped out the build plate and updated my N4, which I hadn’t used in a while – finally got the Carbon in!
Looks like the latest update did the trick: the first layer is incredible!
Can anyone confirm similar results?

closed 2026-10-30 by halfmanbear