OpenNeptune3D/OpenNept4une #279: Webcam install broken?

jubjubrsx, 2025-03-29:
following the instructions I get this? would love some help please

[image]

`==========================================================
Webcam Setup Installation
==========================================================
Do you want to configure mjpg-streamer? (Y/n):  y

Running Webcam Setup Installer...

How many webcams do you wish to configure? (1 or 2)
Enter 1 or 2: 1

Launching crowsnest uninstaller as the standard user...
crowsnest - A webcam daemon for multiple cams and stream services.

        Ahoi!
        To sad that you want to uninstall crowsnest :(
        This will take a while ...
        Please reboot after uninstallation has finished.

Do you REALLY want to remove existing 'crowsnest'? (y/N) Ny

You answered 'Ny'! Uninstall will be aborted...
GoodBye...

make: *** [Makefile:22: uninstall] Error 1

Something went wrong during uninstallation! Please try again...
Webcam Setup Installer encountered an error.
==========================================================
Press enter to continue...`

jubjubrsx, 2025-03-29:
side note webcam worked after unplugging and reseating.... but script still errors?

bvaerewyck, 2025-03-30:
this is what you selected **Ny**
Do you REALLY want to remove existing 'crowsnest'? (y/N) **Ny**
try a back space at that line, and then hit "y" and then enter.