OpenNeptune3D/OpenNept4une #280: Printer not exiting file

F2g-git, 2025-03-29:
When my printer is done printing recently it does not close the file but keeps it open until i run SDCARD_RESET_FILE.
I thought about adding that to the end of print macro. but wanted to address i here first.

I am running the latest version and did not touch my printer.cfg file.

halfmanbear, 2025-04-14:
Hello, could you please attach your klippy.log to help find what's going wrong.

mtmello2003, 2025-05-14:
I had the same issue, and it can happen due to custom start or end G-code in the slicer (in that case, the end gcode). You're probably using a macro that doesn't exist in the full version of Klipper.

By the way, I agree that the error could be handled better. :)

closed 2026-10-30 by halfmanbear