OpenNeptune3D/OpenNept4une #285: No .bin file generated after performing Flashing / Updating Klipper MCUs first 5 steps

kaichi1342, 2025-04-18:
2025-04-18 

No .bin file generated after following all the steps until the Flashing MCU. 
No bin file to download to be able to proceed to the next step.

The firmware folder is empty after the opennept4une -> 4 -> 4 steps and due to this i cant proceed with the installation. 

The above happen with an updated klipper (v.0.13.0-33) as the instruction provided and validated via kauh. 

I also attempted it without updating klipper and it update  to the latest  (v.0.13.0-33) from (v.12.0-32) at the flashing mcu step anyways.

halfmanbear, 2025-04-18:
Thankyou for raising this, it appears the make menu config of the latest klipper (0.13) requires additional configurations.

I've updated the dev branch to include these. 

Could you please switch to the dev GitHub branch within the advanced page of the opennept4une script on the printer via SSH and try again.

It will be helpful feedback for the fix.

Cheers

kaichi1342, 2025-04-18:
Im here to confirm that after switching to dev the .bin files are now available.

closed 2025-04-19 by halfmanbear